cahirwpz/mimiker #3: Integrate newlib with the kernel.

cahirwpz, 2015-12-01:
It's desired to use standard C function within kernel (like `strings.h`, `stdlib.h` and so on) - there's no real benefit in implementing them by ourselves.

In the future It'd be even better if we could devise kprintf function for debugging purposes. Obviously I expect this function to reuse some code from newlib.

rafalcieslak, 2015-12-01:
Is the goal to properly integrate newlib by modifying its autotools scripts so that it can automatically building into an object file that we would then simply link with the kernel, or would it be fine to just copy the appropriate source files from newlib into our source tree to keep it buildable with a simple makefile?

I believe the second approach is probably easier to implement, it significantly simplifies the build process, and should keep our sources self-contained and therefore easier to comprehend as a whole. On the other hand, the first option makes it easier to replace newlib with a newer version or a different implementation (do we need that?), and keeps our source independent from stdlib implementation.

cahirwpz, 2015-12-02:
It's even more complicated that you may think. In fact we need two C standard libraries: one for kernel (may be incomplete) and one for user space. In first case it's acceptable we partially merge sources with our kernel - thus the license of donor library should be permissive. In second case - I guess that we'd like to have a reasonably complete solution to enable running existing software on top of our operating system. It's quite obvious we'll have to provide our method to invoke system calls, hence we should fork newlib and patch it to meet our requirements.

I'm quite convinced we should choose newlib for our user space C standard library. However I have no clear conclusions regarding kernel space one. Perhaps PDCLib?

Please read following article: http://wiki.osdev.org/C_Library

cahirwpz, 2015-12-06:
Perhaps we could just steal [_PDCLIB_print.c](https://github.com/blanham/PDCLib/blob/master/functions/stdio/_PDCLIB_print.c) file to have our own printf implementation? I believe most interesting functions from C standard library could be delivered by [libgcc](https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html#Other-Builtins)

rafalcieslak, 2015-12-09:
I have given PDCLib a closer look. Apparently the glue layer is very small, and almost all functions can be stubbed, therefore it should not require much effort to port the entire PDCLib, and then gradually enable the "advanced" features once implemented.

The build scripts for PDClib are not universal and it might be tricky to adapt them to our needs, but since we are fine with merging the library's sources into the kernel (PD license), we can ignore these scripts altogether.

I have also noticed that PDClib makes some use of global constructors, if that would pose an issue, we manually call these designated functions.

I will experiment with PDClib a bit and share results.

rafalcieslak, 2015-12-10:
I've had success porting PDClib. I have been able to compile and link it without stripping any functionality, and in retrospect it did not require much effort.

In my experiment, I've manually built PDClib into a static library, and then linked the kernel with it. I was then able to call functions like `strncpy` or `printf` from `kernel_main`.

Here are some observations I found interesting:

- To compile the full standard library, a lot of system function implementations had to be stubbed. This applies especially to library functions which make little sense at kernel-level, like `abort`, which normally would send SIGABRT to the calling process. PDClib does that by using `raise(...)`, which checks if there was a "signal handler" set with `signal(...)` and uses `exit(...)` if there was none. `exit`, in turn, is an unimplemented stub and all it does is `errno = ENOTSUP`. So eventually this whole chain of calls is not any useful. In this case, we might want to hook `exit` to a custom kernel `panic` or any other function which would report abnormal termination. However, there are many other functions that will need to stay stubbed before their functionality is actually implemented, these include filesystem operations (`openf`,`readf`,`writef`,`closef`), `time`, `getnpages` and `freenpages` (used by `malloc`), synchronisation support (mutex and conditional variable methods), `system`, `gettmpfile`, `getenv`.

- The synchronisation features currently may have a null implementation (so that a mutex is a `char` and `mtx_lock(mtx_t *mtx) {(*mtx)++;}`). This is fine before we use kernel threads, though at some point this will have to be improved into disabling interrupts, spinlocking etc.

- The `flash` memory area is far too small to fit the library - it has about 64kB. If I understand correctly, the 2MB `prog` area is the actual flash memory of the chipset - if I am right, then it might make sense to move the `.text` section there (but not `.exception`).

- I tried to measure how much does the `.text` section grow when I pull `printf` and its dependencies from the compiled library. I hope I have checked it incorrectly, because I've measured more than 500kB, which would be 1/4 of available flash storage just for the `printf`! I will give this size a closer look, and compare `-O0` with `-Os`.

- I might have stumbled upon a compiler bug where the value of `x` after `if(*(++x) == 'a') {return 0;}` is not incremented by the generated machine code. However, I have been investigating it after compiling with `-Os`, so the debugger output might have mislead me. I will experiment with this issue.

- PDClib uses a single global constructor which initializes locale structures and stdio mutexes. Calling it manually before using any PDClib function is a valid workaround.

- The library headers do some feature checking, but they are very inconsistent - under many configurations some headers will disable a feature, while others will assume it is available. The only variant that I found where all features were consistently detected can be forced by globally setting preprocessor variable `_POSIX_C_SOURCE` to `200809L` (this probably enables everything possible, so we won't miss on any functionality).

- PDClib does not compile without warnings. If we don't want to focus on fixing PDClib, we should ignore them. Also PDClib requires `c99` or `c11`.

- It is very easy to hook stdout to serial port. The `writef` function just has to check whether the file descriptor is 1, and in such case it should pass the bytes to uart_raw driver.

- I needed to link with libgcc. This is because some (floating-point?) operations generated calls to `__udivi3` and `__umodi3`. No other symbols had to be pulled from libgcc.

My current PDClib port is just a quick sketch, so it's not really worth sharing at the moment. In the upcoming days I will clean it up a lot. I've been given confirmation that #6 will be accepted soon, so I'll take that branch as a starting point (it's nice to have a working `.data`). Hopefully in a few days I will share a nice working demo.

rafalcieslak, 2015-12-10:
A closer examination of linker output explained why the `.text` section grew so big: it does indeed take ~500kiB, out of these only ~60kiB are actual code, and ~380kiB is taken up by unicode character data table stored in `.rodata`. It is not trivial to get rid of it, because of tangled dependencies, but I will investigate if we could get rid of that array.

rafalcieslak, 2015-12-10:
I have pushed a working demonstration to branch `pdclib1` (021d480).

cahirwpz, 2015-12-15:
Have you considered using parts of smallclib provided by Imagination Technologies?

The sources are published on very liberal licence (BSD-like) and available after toolchain's installation in `Toolchains/mips-mti-elf/2015.06-05/src/smallclib`.

rafalcieslak, 2015-12-15:
We already use `Toolchains/mips-mti-elf/2015.06-05/mips-mti-elf/include/mips/m32c0.h`! Look at the first line of `startup.S`. The path `Toolchains/mips-mti-elf/2015.06-05/mips-mti-elf/include/` is in the standard include directories list, so it's the MIPS-specific headers are quite conveniently provided.

I've skimmed through smallclib, and it looks like a literal mix of newlib and musl, though significantly stripped. It is much richer than pdclib, implementing much more functions. While some of them will be of little use in the kernel (math functions), there are some that may come in handy, including `setjmp\longjmp`, global constructor and destructor calling (with `.init_array`), some kind of hash tables and various hashing procedures.

smallclib does not provide it's own header files set, it uses the one that comes with the toolchain, so we would have to use the standard include directories.

Also, there is no explicit glue layer in smallclib, it probably assumes some functions (e.g. `write`, I suppose) are already implemented. It never does any manual syscalls, so I expect implementing symbols that linker found missing will be enough to port smallclib to our kernel.

I am already working to provide a demo. Since I'll be explicitly picking (some) functions by hand, it may take me a bit more time than it did with pdclib.

rafalcieslak, 2015-12-15:
Having successfully ported smallclib, I have to admit I like it more than pdclib.

A working demo is available in `smallclib1` branch.

Here are some of my observations:

- The directory layout of smallclib is so much simpler than pdclib's. There is no illusion of non-exisiting plugin system, and all files - strictly one per function - are placed within a directory structure of depth 1. This makes reading implementation so much easier.
- It is easier to pick some functions from smallclib and ignore others, apparently the dependencies between functions are kept to minimum. I chose to build `*printf` functions, most useful string\memory functions, `is*` functions and some useful utilities (eg. `strerror`, `atoi`, `setjmp`), but expanding this selection will be very easy if we need to use others.
- The glue layer is very thin! Using the aforementioned subset of smallclib only had to implement two functions (`write` and `lseek`). Compare that with more than 20 files needed to correctly link pdclib.
- smallclib does not use any global constructors (it does, however, provide helper procedures for calling functions from `.init_array`)
- I was able to compile the standard text-operating functions without compiling unicode and wide character support. Although `snprinf` would like to use wide-char related functions, which in turn pulls whole locale management as a dependency, I was able to trick it by providing a custom stub for `wctomb` which always fails. Unless we use wide characters, that stub should never be called anyway.
- In the end, I tested the library using the same demo procedure as in pdclib's case. Here it is, for completeness:
```
    // Simple printf.
    printf("=========================\nPrintf is working!\n");
    // Both stdout and stderr write bytes to UART1.
    fprintf(stderr,"Stderr working too!\n");

    // Formatting
    printf("This is a number: %d - %x\n", 123456, 123456);
    // String rendering
    const char* stringA = "This is an example string!";
    char stringB[100];
    sprintf(stringB, "Copied example: %s",stringA);
    puts(stringB);

    // String functions:
    printf("Above text has length %zu.\n", strlen(stringB));
    printf("Word: \"example\" is at: %ld.\n", strstr(stringB,"example") - stringB);
    memset(stringB,'a',20);
    puts(stringB);

    // The limits are defined
    printf("INT_MAX is %d\n", INT_MAX);
```

The size of `.text` section for the entire kernel with this example: 24kB (without size optimisations enabled). A half of that size (12kB) was taken up by the sprintf's format interpreter. For comparison, the `.text` section using pdclib has ~500kB.
 - I am also under the impression that smallclib is more standard-conforming than pdclib, I've seen several functions with non-standard return type in pdclib, and no such in smallclib. It also has the advantage of staying fully integrated with the toolchain - it borrows the featureset from the toolchain's system include files.

Given these facts, I suspect we may be leaning towards using smallclib as the kernel's standard library.

cahirwpz, 2015-12-16:
Would it be possible to just use the library from the toolchain instead of importing the whole source tree?

I think one could define `kprintf` function in same way as [`sprintf`](https://github.com/cahirwpz/wifire-os/blob/smallclib1/smallclib/smallclib/stdio/sprintf.c#L66) is defined. Then, you wouldn't have to override `write` and possibly `lseek`.

Having read all your observation I'm feeling convinced that the idea of using smallclib is sound. Let's try to integrate it with the kernel pulling in as little as possible. After that let's quickly move to other important tasks ;-)

rafalcieslak, 2015-12-16:
I was very careful with using the library included in the toolchain, as it might be slightly tricky to get right.

First thing is that in such case a full out-of-tree build is necessary, because the library source directory has to stay clean. Conceptually, this not a problem, but it's not trivial to do that with `make`, because it would need to recreate directory structure, which is painful as any conditional logic within a Makefile. We have, however, the advantage of having all translation units named differently, so it may be simpler to just drop all object files in a single build subdirectory.

The other potential issue I see is the difficulty of locating the library's directory. The path to toolchain's resources is not obvious. The only reliable way of locating the library that I came up with, is to ask the compiler about the location of any standard header, and then assume that the standard library is at a constant path relative to that header's. It is not a very elegant solution, though. I am aware of the environmental variable `MIPS_ELF_ROOT`, but I am unsure whether it can be trusted to have been set correctly.

Both these issues would have been solved by any meta-build system, but I believe this is not the right time to pick one.

Since we are committed to using smallclib, I will create a PR once I resolve the discussed issue.

closed 2015-12-20 by cahirwpz