cahirwpz/mimiker #7: Initialize TLB and turn on memory translation

cahirwpz, 2015-12-15:
Write `tlb_identity_mapping()` routine in C that will recreate default memory management setup (figure 4-4 from PIC32MZ EC documentation). Use functions described in "Codescape GNU Tools for MIPS Programmer's Guide", chapter 13.

After you turn on memory translation, make sure you get TLB exception when invalid access happens. Then try to create a mapping in user segment (addresses up to 2^31-1) and verify it works.

cahirwpz, 2015-12-30:
http://nxr.netbsd.org/xref/src-freebsd/sys/mips/include/tlb.h

rafalcieslak, 2016-01-06:
What exactly do we mean by an identity mapping? Do we want to map `kseg0` and `kseg1` both to `0x00000000-0x20000000`? This would be unnecessary, would waste precious TLB space, and would be not testable, because entries describing `kseg0` and `kseg1` will be always ignored. The PIC32MZ EC documentation apparently does not mention that these virtual address ranges are not mappable, and will be always translated to `0x00000000-0x20000000` regardless of TLB state (and thus we are able to run our current code without setting up TLB) - according to "MIPS® Architecture For Programmers Volume III: The MIPS32® and microMIPS32TM Privileged Resource Architecture".

The TLB initialisation that is performed by The MIPS Boot Example™ simply resets all entries to zero. The same is done by LiteBSD.

rafalcieslak, 2016-01-07:
Furthermore, I believe that in an elegant solution of the mapping problem in general, the TLB would be updated only by the TLB miss exception handler. The handler has to be super efficient, so it should not perform any complex calculations, and just lookup for the requested entry in a global array containing the page entry (the page table would be prepared beforehand). The procedure for creating a mapping would then refrain from accessing TLB at all, and would just insert the mapping information into the page table, so that it can be pushed into TLB by the exception halder, when an underlying address is accessed.

Therefore I believe it may turn out that we don't really need to expose the intefrace for inserting entries into TLB, as mappings should be instead written into the page table.

My point is that we may want to implement basic TLB refilling simultanously with a simple page table, as the implementation of the first would not be useful without the other.

cahirwpz, 2016-01-07:
Please read *"Memory Systems: Cache, DRAM, Disk"* chapter 31.1.3, *"Bottom-up traversal"* section. It describes one of the methods to implement TLB miss handler for MIPS architecture.

cahirwpz, 2016-01-08:
Could you analyze hardware memory map for PIC32MZ and publish it here? Each entry should have following format: `(start_addr, end_addr, attributes, description)`.

rafalcieslak, 2016-01-09:
I generally like the idea of traversing the page table bottom-up, but I do not understand what problems does it solve (except for using one load less, which does not sound like a big deal). I also suspect that it may be noticeably harder to implement than top-down traversal.

Also, unless I got it wrong, each process still needs to come with it's own root page table, so it may make sense to compress it further by using even less than 10 bits for addressing it, as allocating 4kB per process may be costly with only 512kB of RAM.

I'd be happy to provide such listing of hardware memory map, but I am not sure how much detail should I describe, because, honestly, I do not understand the purpose of such list. I believe such map is already provided in *"PIC32MZ EC Family Data Sheet"* chapter 4.1. What else, exactly, do we need to know to implement the TLB miss handler?

cahirwpz, 2016-01-17:
Perhaps the implementation of [TLB miss handler](http://fxr.watson.org/fxr/source/mips/mips/exception.S#L120) from FreeBSD will guide you how to organize page table.

coodie, 2016-04-10:
In order to be able to use these functions we need to link the project with -lhal (section 5.8.3 of the Codescape GNU Tools Progammers' Guide). The problem is there are some linking errors. Some of them can be easily fixed by copying missing PROVIDES from uhi32.ld, but there is also a problem with other stuff. This can be fixed by playing with the code, but of course this might mess up something we already have. I think another alternative would be to write some of these functions in assembly.

cahirwpz, 2016-05-13:
Merged with #50.

closed 2016-05-13 by cahirwpz