cahirwpz/mimiker #13: The .elf file rule does not depend on smallclib

rafalcieslak, 2015-12-21:
Steps to reproduce:

1. Create a clean copy of repo, or use `make clean`.
2. Run `make qemu`
3. The linker will complain that it cannot link the .elf file because `smallclib.a` is missing.

Using `make all` to build the .elf file will, however, succeed, because the `all` target depends on both `smallclib` and `.srec` file, and they happen to be build in the correct order.

Note: This issue was introduced by 870cbfdadff3f713a6c4f76c54bf963cb7a5a652

Now that we have a generic rule for building any .elf file, I sincerely have no idea how to add the missing dependency into Makefile.common. Requiring `smallclib.a` is not a universal property of all .elf files, so it probably should not be added to the generic .elf pattern rule.

cahirwpz, 2015-12-21:
It seems to be easy fixable with following workaround:

$(PROGNAME).elf: smallclib $(OBJECTS) $(LDSCRIPT)

What do you think?

rafalcieslak, 2015-12-21:
This would work. However, in such case the .elf rule would no longer be generic. It would only make sense in the `./Makefile`, as any other Makefile has no `smallclib` rule defined. Therefore I believe it would make sense to move the .elf rule from `./Makefile.common` back to `./Makefile`, as it becomes specific.

closed 2016-01-17 by cahirwpz