cahirwpz/mimiker #18: Generic linked list implementation.

cahirwpz, 2015-12-30:
We need it badly, now. Two major requirements are:
* no memory allocation for list nodes (i.e. nodes are embedded into structures),
* any structure may be easily linked by several lists. 

There're a few solutions:
* [Linux Driver Development](http://www.makelinux.net/ldd3/) chapter 11.5.
* [FreeBSD queues](http://nxr.netbsd.org/xref/src-freebsd/sys/sys/queue.h)

Just choose bare minimum interface and present it as a header file, don't start implementation yet. I'd go for doubly linked circular list variant - it seems to be the most versatile.

Killavus, 2016-01-04:
I just created the pull request being the first minimal step to implement this feature.

cahirwpz, 2016-01-11:
We need those pesky lists now - please prepare a patch ASAP :)

Killavus, 2016-01-12:
It'll be done today :)

cahirwpz, 2016-01-15:
I'm still waiting...

closed 2016-01-16 by cahirwpz