cahirwpz/mimiker #25: The linker incorrectly sets _estack

rafalcieslak, 2016-01-09:
I noticed this issue while implementing a bootstrap memory allocator, but it can certainly haunt us in any context.

Our intention is to place the kernel stack at the end of the RAM (as it grows downwards). This is expressed in `pic32mz.ld` with:

    _estack = ORIGIN(ram) + LENGTH(ram);

This value apparently makes sense, `pic32mz.map` confirms:

                    0x00000000a0080000                _estack = (ORIGIN (ram) + 0x80000)

which is, indeed, 512kB after RAM's beggining. This address value is correct.

To initialize stack pointer, we use the `la` pseudo-instruction: (`startup.S`)
<pre>
_start:
    <b>la  $sp, _estack</b>    // Set stack pointer.
    la  $gp, _gp        // Prepare global pointer.
</pre>

I would expect that the linker will set the value in this instruction so that the effect of this instruction will be setting `sp` to `0xa0080000`. However, it is not:

<pre>
(gdb) disas _start
Dump of assembler code for function _start:
   0xbfc00500 <+0>:	 lui	sp,<b>0xa000</b>
   0xbfc00504 <+4>:  addiu	sp,sp,<b>4236</b>
   0xbfc00508 <+8>:	 lui	gp,0xa000
   0xbfc0050c <+12>: addiu	gp,gp,0
   [...]
</pre>

so in fact (in this case) the stack pointer will be set instead to `0xa000108c`. And it indeed is:

<pre>
(gdb) info registers
[...]
            t8       t9       k0       k1       gp       sp       s8       ra
 R24  00000000 00000000 00000000 00000000 00000000 <b>a000108c</b> 00000000 00000000 
[...]
</pre>

Obviously, this is incorrect. It will cause all sorts of issues, starting with the fact that the stack ends up having just 1kiB before it crashes onto `.bss`. I originally found this issue when I tried to allocate more than 1kiB with the bootstrap memory allocator, and cleared the allocated region, which happened to override stack frames.

I have no idea what may be causing this. The only interesting thing I noticed while investigating this is that apparently that incorrect address is always `_end + 1kiB`, so increasing `.data` or `.bss` by N bytes pushes the stack these N bytes further.

rafalcieslak, 2016-01-09:
Another interesting find is that if I remove the `_estack` definition from the linker script, the kernel still compiles and links successfully. Apparently this symbol is already defined elsewhere. GNU ld docs do not mention anything here, so I will assume it is the imgtec's toolchain that introduces and prefers a different definition of `_estack`. Indeed, renaming `_estack` solves the issue:
<pre>
disas _start
Dump of assembler code for function _start:
   0xbfc00500 <+0>:	lui	sp,<b>0xa008</b>
   0xbfc00504 <+4>:	addiu	sp,sp,0
   0xbfc00508 <+8>:	lui	gp,0xa000
   0xbfc0050c <+12>:	addiu	gp,gp,0
</pre>

<pre>
(gdb) info registers
[...]
            t8       t9       k0       k1       gp       sp       s8       ra
 R24  00000000 00000000 00000000 00000000 00000000 <b>a0080000</b> 00000000 00000000
[...]
</pre>

I will prepare a fix in a sec.

rafalcieslak, 2016-01-09:
Actually, this is not toolchain's fault. **We** provide a conflicting `_estack_` definition. The correct one is in [`pic32mz.ld:18`](https://github.com/cahirwpz/wifire-os/blob/0a218d76f98b96f3d5465254c71d91ecf4716cb3/pic32mz.ld#L18), the conflicting one is in [`startup.S:74`](https://github.com/cahirwpz/wifire-os/blob/0a218d76f98b96f3d5465254c71d91ecf4716cb3/startup.S#L74). Removing the incorrect one fixes the issue.

cahirwpz, 2016-01-10:
Oh... wow. It really was defined twice. Good job! :+1:

closed 2016-01-10 by cahirwpz