cahirwpz/mimiker #31: Physical page allocator

cahirwpz, 2016-01-12:


coodie, 2016-02-11:
We have already discussed it in real life but for purpose of keeping good documentation and everything visible to everyone I post here proposed interface:

`struct vm_page{
    vm_paddr_t phys_addr;
    vm_paddr_t virt_addr;
    vm_page *link;
    size_t size;
}

void vm_phys_init();
void vm_phys_add_seg(vm_paddr_t start, vm_paddr_t end)
vm_page* vm_phys_alloc(size_t order)
void vm_phys_free(vm_page* page)
`
Annotations:
`vm_phys_alloc` takes order and returns page of size 2^order.
`vm_phys_add_seg` takes segment to be managed by the system. The system manages couple of segments, allocating memory from any of them.
`link` and `virt_addr` might not be useful at the moment but will be useful later

Of course this might change during implementation, but idea stays roughly the same.

cahirwpz, 2016-02-11:
Ok. I updated my interface proposal. Could you elaborate on `vm_phys_seg.free_queues` ? I'd like to know how would you implement data structures for buddy system algorithm.

coodie, 2016-02-11:
Well now i see other problem. My idea would be to keep sorted list of free segments and then split them and coalesce on demand. Also to move them around when needed. For example: 

I need segment of order 2^5, but i have none, however i have segment of size 2^7, so i split it into two segments of sizes 2^6, and then one of 2^6 segments i split into two segments of size 2^5. 
So i started with {2^7}, and now i have {2^6, 2^5} (because i allocated one 2^5 segment).

This means that general number of items collected in the array would **change in time**, which means i'd need to use allocator in order to maintain it optimally. So the idea would be not to use list... but array with max_size. This also is a problem because this takes tons of memory and is simply wrong. 

I'll do some search in implementation of FreeBSD and see how it's actually done.

 Here is similar struct from free_bsd (can be found in vm_phys.h)

```
struct vm_phys_seg {
    vm_paddr_t      start;
    vm_paddr_t      end;
    vm_page_t       first_page;
    int             domain;
    struct vm_freelist (*free_queues)[VM_NFREEPOOL][VM_NFREEORDER];
};
```
vm_freelist is implemented like this:

```
struct vm_freelist {
    struct pglist pl;
    int lcnt;
};
```

pglist is implemented like this:

```
TAILQ_HEAD(pglist, vm_page);
```

Above means that pglist is list of type vm_page.

cahirwpz, 2016-02-12:
Ok... firstly let's settle on terms semantics. Don't use "segment" word to describe page of size of 2^n.

As for dynamic allocation, I believe that adding `TAILQ_ENTRY(vm_page) freelist` to `vm_page` solves the problem. You'd have then a storage area to link all free pages of the same size.

cahirwpz, 2016-02-12:
Ok. I think my current proposal is quite complete. Can you verify that?

coodie, 2016-02-12:
This looks (almost) fine. 

What with vm_page_array i mentioned in previous post? And what are semantics behind vm_phys_add_seg ?

cahirwpz, 2016-02-12:
With regards to `vm_phys_add_seg`, how are you going to create initial list of `vm_phys_seg`'s and populate it with data? As with `vm_page_array`, please look at the field `vm_page_t page[0]' in `vm_phys_seg` IMHO that should solve the problem.

coodie, 2016-02-13:
The point is, i don't see reason to use separate memory segments, and add them on demand since we have one big contiguous memory segment which's size is 512k. 

It would be easier to prepare array of all pages of const size. Something like:

```
struct vm_page vm_page_array[MEMORY_SIZE / PAGE_SIZE];
```
It's size is constant and instantly allocated in .data segment while compiling. I'd like to keep one, global list of segments like:

```
struct vm_phys_seg basic_seg; 
```
Everything is going to land in .data segment. No excess memory use, and algorithm will just manipulate links in `vm_page_array`. That's how it's done in `free_bsd`.
`vm_phys_init` will then check some addresses and set up initial memory segment. This is similar to:

```
void vm_phys_init(){
    basic_seg.start = RAM_START + DATA_SIZE;
    basic_seg.end  =  RAM_START+RAM_LENGTH - MAX_STACK_SIZE
   
   ... setting up queues ...
 
}
```

cahirwpz, 2016-02-13:
I think we've already spent too much time talking about `vm_phys_add_seg` issue - I don't find it productive to muse on it. Do it as you wish, I'll rework the PR to my taste anyway.

closed 2016-04-05 by cahirwpz