cahirwpz/mimiker #32: Interrupt handler registration and chaining

cahirwpz, 2016-01-12:


coodie, 2016-02-18:
Before i start digging the topic. Do you know if this implementation on wifire would differ greatly from the malta we'd like to port it to? I'd like to know if there is a point in doing this, before i (might) realise i just wasted time.

cahirwpz, 2016-02-18:
Qemu doesn't implement vectored interrupts with non-zero offsets. Thus we configure interrupts in almost the same way as on original MIPS platforms. There're some minor differences but I strongly believe the rest will be the same.

coodie, 2016-03-05:
Keep in mind that don't understand problem well, that's why i'm starting discussion.

What is purpose of filter handler function? As far as I understood `freebsd` code, filter function is the first one to be executed when interrupt comes, it's name is filter, because some interrupts are stray (what is stray interrupt and when can it occur ?), and it filters them in case. Filter is executed in interrupt routine so this cannot use some complicated synchronisation methods (I don't know why, but I have some intuition), that's why some work has to be deferred to threads which can basically do everything. A little bit about this here:

https://lists.freebsd.org/pipermail/freebsd-hackers/2009-January/027552.html

Since `freebsd` uses special threads to handle interrupts (and we don't have threads yet), is there a point in making this task now, before we have threads? Without threads we'll have to restrict ourselves from using some functionality (which doesn't exist at the moment, but it will).

What kinds of interrupt would you want to handle right now? Not every interrupt is IRQ based (like exceptions for example), there is 190 of them on pic32mz, and we mostly care about clock interrupt at the moment and maybe software interrupts (traps) to make system calls. If we plan to use some simple code in interrupt handlers, then we might not need to use threads to implement this.

cahirwpz, 2016-05-13:
I'll consider this task to be finished when at least two devices connected to single interrupt are correctly handled. To achieve it in the simplest form you should program `RTC` and `UART0` to generate interrupts: https://github.com/cahirwpz/mimiker/blob/master/include/malta.h#L38

cahirwpz, 2016-12-16:
@dyniec There's already an implementation of `intr_handler` and `intr_chain` machine independent functions in `sys/interrupt.c`. Please read `bus_setup_intr(9)` manual.

cahirwpz, 2017-05-21:
Implemented with 2126c0ff04c3c1e29a23b76db5e46770a9d2024c

closed 2017-05-21 by cahirwpz