cahirwpz/mimiker #33: Kernel threads

cahirwpz, 2016-01-12:


cahirwpz, 2016-05-14:
First working version should allocate `thread_t` structure using `kmalloc` and stack using `vm_phys_alloc`. I expect two threads to print out, using serial port, an identifier letter and be switched every 10ms. After several second they should call `thread_exit` and main thread should continue after it `thread_join`'ed both child threads. Early locking mechanism may be necessary to handle aforementioned functionality.

Moreover, first path of execution (i.e. one that enters `kernel_main`) has to be converted to a thread. That will require stack switching mechanism.

cahirwpz, 2017-03-24:
This is almost ready to be closed. We need #185 to be completed.

rafalcieslak, 2017-03-24:
I'll reapply #185 onto master updating it to current interface, fix synchronization, and create a new PR.

closed 2017-04-07 by cahirwpz