cahirwpz/mimiker #36: Task queue

cahirwpz, 2016-01-17:
A task queue is needed to run bottom halves of interrupt handlers. A top half of interrupt is executed during interrupt request handling. Time consuming and possibly blocking work is deferred to be executed within high priority kernel thread.

```
typedef void work_fn_t(void *data);

struct work {
    STAILQ_ENTRY(work) work;
    work_fn_t *func;
    void *data;
};

Killavus, 2016-01-17:
So what we basically want to do is to accept every intr handler asap, but if it needs more work we'd like to keep such work within a queue and process it later (if kernel is busy)?

I'd like to take a look at how BSD does it. I believe "your data structures" means "steal it if it is possible and feasible for us"? I'd rather want to minimize the 'talk part' and focus on delivering it since I failed with list implementations hard.

cahirwpz, 2016-01-17:
Nope. Please read the justification behind splitting interrupt handler into two parts. I hoped it had been well described during a seminar talk. There're several good articles about that. 

Research and discussion is very important - don't neglect it. I have no time to investigate every implementation... you should convince me that stealing / implementing one particular solution is a good idea.

Killavus, 2016-01-17:
OK. I'll refresh the talk from seminar (because I just remembered this particular part) and will research on the topic. Thanks!

cahirwpz, 2016-02-08:
Any progress?

cahirwpz, 2016-02-12:
The scope of this task is unclear to me. It has to be redefined from scratch.

closed 2016-02-12 by cahirwpz