cahirwpz/mimiker #37: Generic kernel memory allocator

cahirwpz, 2016-01-18:
TBD

cahirwpz, 2016-02-16:
The implementation may use following:

```
typedef struct mem_block {
    TAILQ_ENTRY(mem_block) mb_freelist;
    size_t size;
} mem_block_t;

typedef struct mem_arena {
    TAILQ_ENTRY(mem_arena) ma_list;
    size_t pages;
    TAILQ_HEAD(, mem_block) ma_blocks;
} mem_arena_t;
```

Darge, 2016-05-06:
1. So we do want a realloc?

2. When one uses `malloc()`, we iterate through all the mem_arenas, inside which we iterate through all the free memory blocks and find the first fit. Is that correct?

3. How exactly should `MA_THRESHOLD` work? And what is the reasoning behind this?

4. I think that when describing some continuous memory, a pointer indicating the beginning of the memory and its size is more natural than two pointers. Do you agree?

5. We have to make sure that mb_data is properly aligned. Without doing that, the compiler will think that alignment isn't required, because mb_data's elements are bytes. If we later allocate some space for an integer and assign something to that integer, the compiler will generate a "store word" instruction, which fails if the address is misaligned. <br>
To achieve proper alignment, this:
`uint8_t mb_data[0];`
should become this:
`uint64_t mb_data[0];`<br>
I suppose that 4 bytes aren't enough (even though this is the word size) because doubles are 8 bytes. I will have to confirm this though.<br>
You can check for yourself that these two structs have different sizes:
`struct aaa
{
  char y;
  char data[];
};
`
`struct aaa
{
  char y;
  double data[];
};`

Btw I have learned about number 5 the hard way.

cahirwpz, 2016-05-06:
1. Ignore realloc for now. I do not find it useful yet.
2. Yes. Iterate all arenas and use first-fit inside. Later we will add other placement policies if such need arises.
3. Ignore it for now again. We will use malloc for small structures.
4. When you're about to free a block you need to check if block's pointer belongs to the arena. This is not really important - we can change it later quickly, if we recognize one way is better than the other.
5. You're right! Alignment rules work that way! I.e. choose as the alignment the size of largest machine type constituent of a structure.

Darge, 2016-05-07:
Should the two pointer parameters be aligned as a part of the interface requirements, or should the function `malloc_add_arena()` handle that?

Handling that inside the malloc will add some complexity to the code, because we can't even put the `mem_arena_t` structure anywhere - it has to be aligned too. Definitely not fun.

cahirwpz, 2016-05-08:
1. I assume that user will pass a couple of pages to `malloc_add_arena`.
2. I'm not 100% sure, but I guess it won't hurt for now if I say yes.
3. `mem_block` doesn't need any explicit flags field at all. `mb_size` encodes everything we need.

coodie, 2016-05-13:
1. Out of curiosity. How does mb_magic work in detecting programmer's errors? 

2. Current implementation has a compile-time bug. As far as I understood everything I guess I have to use following macro and give it some memory from vm_phys_c. 
```
#define MALLOC_DEFINE(pool, desc) \
    malloc_pool_t pool[1] = {     \
        {{NULL}, MAGIC, desc}     \
    };

typedef struct malloc_pool malloc_pool_t;
```
A careful C programmer will notice that this macro is pretty much useless and will never compile (unless used in malloc.c file), because malloc_pool_t is incomplete struct type. I understand that someone wanted to hide implementation details from user, but I suspect it's quite impossible unless you redefine this macro into function.

cahirwpz, 2016-05-13:
Ad.1 We hope to detect that it got overwritten as soon as possible. That frequently helps.
Ad.2 You're right. @Darge didn't make sure it works outside of `malloc.c`. I quickly fixed that.

closed 2016-05-25 by Darge