cahirwpz/mimiker #40: Sleep queues

cahirwpz, 2016-02-12:
Develop a mechanism that will have similar responsibility to FreeBSD's [sleepqueue](https://www.freebsd.org/cgi/man.cgi?query=sleepqueue&sektion=9). For reference look at: [sleepqueue.h](http://fxr.watson.org/fxr/source/sys/sleepqueue.h) and [subr_sleepqueue.h](http://fxr.watson.org/fxr/source/kern/subr_sleepqueue.c).

> Sleep queues provide a mechanism for suspending execution of a thread
> until some condition is met.  Each queue is associated with a specific
> wait channel when it is active, and only one queue may be associated with
> a wait channel at any given point in time.

```
typedef struct sleepqueue {
  TAILQ_HEAD(, thread) sq_blocked;  /* Blocked threads. */
  uint32_t sq_blockedcnt;           /* Number of blocked threads. */
  void *sq_wchan;                   /* Wait channel. */
} sleepqueue_t;

typedef struct sleepqueue_chain {
  LIST_HEAD(, sleepqueue) sc_queues;  /* List of sleep queues. */
} sleepqueue_chain_t;

void sleepq_init();

/* Places the current thread on the sleep queue for the specified wait channel. */
void sleepq_add(void *wchan);

/*
 * Look up the sleep queue associated with a given wait channel.
 * If no queue is found, NULL is returned.
 */
sleepqueue_t *sleepq_lookup(void *wchan);

/* Find the highest priority thread sleeping on a wait channel and resume it. */
void sleepq_signal(void *wchan);

/* Resume all threads sleeping on a specified wait channel. */
void sleepq_broadcast(void *wchan);

/*
 * Resumes a specific thread from the sleep queue associated with a specific
 * wait channel if it is on that queue.
 */
void sleepq_remove(thread_t *td, void *wchan);
```

cahirwpz, 2016-05-14:
Please use [include/sleepq.h](https://github.com/cahirwpz/mimiker/blob/master/include/sleepq.h) as a starting point, please.

Darge, 2016-05-25:
How will we keep the pointer to the current thread structure?
Linux calculates its address using the stack pointer: [link](http://stackoverflow.com/questions/12193281/how-does-the-kernel-know-what-is-the-current-thread).

cahirwpz, 2016-10-10:
Your changes were merged after minor fixes with master. Nonetheless development of sleep queues must continue as the module is incomplete. Thanks for your work!

closed 2016-10-10 by cahirwpz