cahirwpz/mimiker #41: Basic scheduler

cahirwpz, 2016-02-12:


Darge, 2016-02-13:
Clarification: Should this scheduler be preemptive? Should it be executed during interrupts? Only during interrupts?

Right now, we have two context switch routines: [one for when an interrupt comes](https://github.com/cahirwpz/wifire-os/blob/master/intr.S), and [one executed manually](https://github.com/cahirwpz/wifire-os/blob/master/context-mips.S) via a function call.
Currently, when an interrupt shows up, the interrupt handling routines save the current context on the stack, execute the interrupt handler, and retrieve the context from the stack. Some of the registers aren't saved on that stack, because they aren't modified throughout the whole interrupt handler. We just leave them on the CPU.

When doing a scheduler-induced context switch, we need to save more registers than the interrupt context switch does.
The reason we have to save more of them is that after handling an interrupt that occured when we were running THREAD1, we may preempt it and decide that it's time to run THREAD2.
This wasn't a problem before, because after receiving an interrupt when running THREAD1, we were always resuming THREAD1's execution.

Here comes the question:
**Should we unify both of these context switches to one?**
1. If we did it, then interrupt handling would become less efficient, as we would save registers we didn't have to. But it would be simpler, because we would have one mechanism only.
2. The alternative is to keep it as it currently is, but then making a scheduler-induced context switch would be cherry-picking registers from different sources: some of the registers of THREAD1 are saved on its stack, and some ($s, $sp) are still on the CPU.
And then we would also have to handle running the scheduler without any interrupt: when some thread blocks/goes to sleep and gives up the CPU voluntarily.

Is that what you meant by [CPU context and thread context](https://github.com/cahirwpz/wifire-os/issues/9#issuecomment-172376938)?

cahirwpz, 2016-02-14:
> Clarification: Should this scheduler be preemptive? Should it be executed during interrupts? Only during interrupts?

The scheduler should be preemptive and some of its functionality must be safely callable from an interrupt context.

> Should we unify both of these context switches to one?

Note, that scheduler context switch procedure is called after timer interrupt handler procedure has been entered. Kernel executing in hardware interrupt handler context may not block! It can only delegate a task to worker queue.

I'm not sure if I understand your question correctly. I'm convinced that interrupt handler entry procedure must dump all registers on interrupted kernel thread stack. After that we have two (only?) cases to consider:

1. Handle hardware interrupt and return.
2. A scheduler kicks in and decides to switch current thread to another. Then we need to use low-level procedure for context restoration. It will be basically the same as second part of `intr.S` but will pick up registers from another stack. Effectively, we forget the context of kernel scheduler procedure.

Don't worry much about performance. The solution has to be correct first.

cahirwpz, 2016-02-16:
You should implement #45 with some basic time routines before attempt to build preemptive scheduler. You'll use callout to wake up scheduler after a timeslice is expired.

cahirwpz, 2016-10-07:
Preliminary work on basic scheduler is finished with 3a989a7aaef2a850042f98b0d3cfe20281960729. Obviously the component needs further development...

closed 2016-10-07 by cahirwpz