cahirwpz/mimiker #42: Virtual memory allocator

cahirwpz, 2016-02-12:


coodie, 2016-05-13:
Here is initial interface we already discussed:

```
typedef uint32_t asid_t;
typedef uint32_t pt_entry_t;

typedef struct page_table
{
  pt_entry_t* rpt; 
  pt_entry_t* upt; 
} page_table_t;

typedef struct vm_map_entry
{
  TAILQ_ENTRY(vm_map_entry);
  TAILQ_HEAD(,vm_page_t) pages;

  vm_addr_t start;
  vm_addr_t end;
} vm_map_entry_t;

typedef struct vm_map
{
    asid_t asid;
    vm_addr_t start;
    vm_addr_t end;
    page_table_t *page_table;
    TAILQ_HEAD(,vm_map_entry) entry_list;
} vm_map_t;

vm_map_t* vm_map_new(asid_t asid, vm_addr_t start, vm_addr_t end);
vm_map_entry_t* vm_map_add_entry(vm_map_t* vm_map, uint32_t flags, size_t length, size_t alignment);
void vm_map_remove_entry(vm_map_t *vm_map, vm_map_entry_t *entry);
void vm_map_attach_entry(vm_map_t*, vm_map_entry_t*, vm_addr_t start, vm_addr_t end, uint32_t flags);
void vm_map_pte_rebuild(vm_map_t*);
```

coodie, 2016-05-13:
I have accidentaly closed this Issue so I had to reopen it again.

I think we don't need to add pmap subsystem. FreeBSD has additionak subsystem for managing physical pages because physical pages are platform-dependent. Since our kernel is MIPS only, this isn't an issue. I guess adding pmap might add some implementation overhead, because this needs discussion, thinking about what should it actually do etc. Anyway, I think that pmap would be very small subsystem which would contain few functions, we could skip this part and integrate pmap with vm_map.

cahirwpz, 2016-05-13:
I agree. For simplicity's sake we shouldn't add an abstraction that we won't benefit from.

Please review my changes to proposed interface.

coodie, 2016-05-13:
I don't see any 'big changes', just those which will make my life easier. I like idea of using bit fields for pte, totally forgot about this nice feature of C. 

I don't think we can discover more possible problems at the moment.

coodie, 2016-05-14:
In some cases the subsystem needs to allocate pages directly from vm_phys. For example when creating vm_map we need to make sure we already have rpt space (thus we need to keep reference to rpt vm_page_t object somewhere to avoid memory leak), and when adding entries we have to update upt which is virtualised and that means we need to add physical pages to it on demand. 

I don't really know when are we going to attach physical pages to vm_map_entries, and how are we going to do this. I suspect this is done lazily by the kernel as there are writes/reads from virtual memory (at the moment this makes most sense to me). This is nice because some vm_map_entry can be used for disabling access to memory e,g : (0,4096) memory address for detecting null ptrs, detecting stack overflows, some extra memory protection, we will never want physical pages to be wasted for this, and the system will know about this. 

So I came up with this question:
When do we attach physical pages to vm_map_entry ?

cahirwpz, 2016-05-14:
You're trying to embrace a concept of `vm_object` which I did not considered so far, because it adds another level of complexity over `vm_map`. I'll be fully content with a solution that always attaches pages explicitely, at least for now. Don't bother with user space yet, we'll get there eventually.

Remember you can always check if a given range of *pte*'s is mapped in *page table* (upt) by looking at *directory table* (rpt). We agreed that *directory table* should always be kept in physical memory.

cahirwpz, 2016-10-07:
Anonymous `vm_object` works so I guess the task is finished, but the development of this subsystem is certainly not finished.

closed 2016-10-07 by cahirwpz