cahirwpz/mimiker #45: Add callout queue to support timeouts.

cahirwpz, 2016-02-16:
Note that some functionality from [sys/time.h](http://fxr.watson.org/fxr/source/sys/time.h) may have to be stolen. For reference look at [callout.h](http://fxr.watson.org/fxr/source/sys/callout.h) and [kern_timeout.c](http://fxr.watson.org/fxr/source/kern/kern_timeout.c)

```
typedef void timeout_t (void *);
typedef int64_t sbintime_t;

typedef struct callout {
	TAILQ_ENTRY(callout) c_link;
	sbintime_t c_time;      /* ticks to the event */
	timeout_t *c_func;  /* function to call */
	void	*c_arg;     /* function argument */
	uint16_t c_flags;
	...
} callout_t;

#define	CALLOUT_ACTIVE		0x0001 /* callout is currently active */
#define	CALLOUT_PENDING		0x0002 /* callout is waiting for timeout */

#define	callout_active(c)	  B_SET((c)->c_flags, CALLOUT_ACTIVE)
#define	callout_deactivate(c) B_CLR((c)->c_flags, CALLOUT_ACTIVE)

void callout_init();
callout_t *callout_setup(callout_t *handle, sbintime_t time, timeout_t *fn, void *arg);
void callout_stop(callout_t *handle);
void callout_process(sbintime_t now);
```

Darge, 2016-05-15:
Done in #48.

cahirwpz, 2016-07-05:
I will consider this task to be completed when callout queue is used to trigger short-term scheduler for threads.

closed 2016-08-09 by Darge