cahirwpz/mimiker #46: Pooled memory allocator

cahirwpz, 2016-02-16:
Please read [pool(9)](http://netbsd.gw.com/cgi-bin/man-cgi?pool+9+NetBSD-current) from NetBSD.

cahirwpz, 2016-05-13:
Bitmapped memory allocator for objects of fixed size.

```
typedef struct pool_item_header {
  vaddr_t  ph_page;      /* this page's address */
  uint16_t ph_nused;     /* # of chunks in use */
  uint16_t ph_start;     /* start offset in page */
  uint32_t ph_bitmap[0];
};

LIST_HEAD(pool_page_list, pool_item_header);

typedef struct pool {
  pool_page_list_t pp_empty_pages;
  pool_page list_t pp_full_pages;
  pool_page_list_t pp_part_pages;  /* Partially-allocated pages */
  size_t           pp_itemsize;    /* Size of item */
  size_t           pp_align;       /* Requested alignment, must be 2^n */
  size_t           pp_npages;      /* # of pages allocated */
  size_t           pp_nitems;      /* number of available items in pool */
} pool_t;

void pool_init(pool_t *pp, size_t itemsize, size_t align);
void pool_destroy(pool_t *pp);
void *pool_get(pool_t *pp);
void pool_put(pool_t *pp, void *item);
int pool_add(pool_t *pp, int nitems);
```

coodie, 2016-08-21:
I think it is good idea to read this awesome and really good quality article about slab allocator:

http://citeseerx.ist.psu.edu/viewdoc/summary?doi=10.1.1.29.4759

cahirwpz, 2016-08-28:
@coodie Could you propose an interface and data structures based on the article you linked?

coodie, 2016-08-29:
At this moment allocation looks like this:
```
vm_map_entry_t *entry = (vm_map_entry_t*)kmalloc(sizeof(vm_map_entry), 0);
vm_map_entry_init(entry);

//do something

vm_map_entry_destroy(entry);
kfree(entry);
```

Goal:
We want memory allocation to be used like this (in case of vm_map_entry for example):
```
vm_map_entry_t *entry = (vm_map_entry_t*)pool_alloc(vm_map_entry_pool);

//do something

pool_free(entry);
```

This is very similar to C++ `new`, but is different because `new` consist of memory allocation and then using specific constructor on allocated memory. In our case we can't use specific constructor, but just one constructor (which is more of initializer rather than constructor), so if we want to do something really complicated with the object at creation time this isn't good interface, but keep in mind that's not it's purpose. If you don't like initializer functions we can skip this part and just allocate fixed-size objects, however I'd rather use them now because we have to initialize everything anyway, and in future, when we have mutexes, we will have to initialize them every time we allocate object, thus it's good idea to get used with new allocation interface.

Slab allocator was introduced to improve performance. We are really far from performance problems, therefore for now I'd like to focus on other features of this allocator which are:
- Better and easier to use interface, once user defined a pool and initializer function, there is no need to call them every time object is allocated.
- Debugging features. Quite easy to check double free, usage of freed memory etc. (all of that is described in section 6 of the article).

Now, that i convinced you to use initializer functions, everything is pretty much in the article.

Basic user interface functions:
```
struct pool_t
{
   //pool implementation details here
   //can be hidden from user
};

pool_t *pool_create(char *name, size_t size, int align, void (*constructor)(void *, size_t),void (*destructor)(void *, size_t));
pool destroy(pool_t* pool);

void *pool_alloc(pool_t* pool, uint32_t flags);
void pool_free(pool_t* pool, void* ptr);
void pool_dump(pool_t*);
```

When it comes to pool_alloc function, I'd rather it returned specific type, so we can avoid obvious casting from `void*` to needed type. This shouldn't be hard to implement using macros, and not really necessary.

I don't want to put more advanced user functions, because I can't predict what will be needed. At the moment I think these would perform some operations on pools, like resizing, adding more objects to pool, management of policies on pool etc.

cahirwpz, 2016-08-30:
Proposed interface looks ok. I have some doubts wrt. `align` argument to `pool_create`. I can't see it being useful for the time being. I'd just apply default alignment (of single cache line) for each pool. Please disregard all cache optimization mentioned in chapter 4, for the sake of simplicity. Please give debugging features enough attention - these are going to be extremely useful.

Before you attempt to implement the interface above I'd like to see description of all data structures that you're going to need.

> Show me your flowcharts [code], and conceal your tables [data], and I shall continue to be mystified; show me your tables [data] and I won't usually need your flowcharts [code]: they'll be obvious.
>
> -- Fred Brooks, "The Mythical Man Month", chapter 9.

coodie, 2016-10-01:
Now that I thinked of it, it is good task for someone else because it is quite independent from other parts of the system. I have pretty much good idea about how slab allocator works, and getting this task done wouldn't give me really any benefits except maybe improving my implementation skills. This task is relatively easy to do, but very valuable for someone not familiar with idea of slab allocator. I could help potential volunteer, or do it by myself If nobody attends to do this.

 I'd rather do something new I'm not familiar with - possibly some of low-level stuff.

rafalcieslak, 2017-04-28:
This feature is being worked on in PR #240.

cahirwpz, 2017-05-20:
Implemented with #240

closed 2017-05-20 by cahirwpz