cahirwpz/mimiker #50: Implement virtual address memory mapping for kernel address space

coodie, 2016-04-21:
We are quite far from using user mode, but virtual address space for kernel is going to be useful. For example, we could make use of memory protection to detect stack overflows (this is impossible without VA). This is going to be easier to implement higher level memory allocator using VA, since buddy system we have returns pages of size of power of two, we might try to create contiguous mappings to these of other sizes which aren't powers of two. 

We also already know how to use TLB on our board. I propose two-level page table as described in _Memory Systems: Cache, DRAM, Disk_ book in chapter 31.1.3 (Bottom-up traversal page table).

Our board has some support for this task. For example there is _Context_ register which contains base address of page table. 

For simplicity, I wouldn't use variable sized pages to be put in TLB (in order to reduce TLB misses by mapping big segments of memory), unless this turns out to be easier than we might expect. Another solution would be to use bigger pages for kernel (like 2MB pages). 

Some references are here:
http://nxr.netbsd.org/xref/src-freebsd/sys/i386/include/pmap.h
http://nxr.netbsd.org/xref/src-freebsd/sys/mips/mips/pmap.c
http://nxr.netbsd.org/xref/src-freebsd/sys/vm/vm_map.h
https://www.freebsd.org/cgi/man.cgi?query=vm_map

I also recommend reading 6.4 of _The design and Implementation of FreeBSD operating system_

This isn't ready yet. I'll update this comment as I gain more information.

cahirwpz, 2016-05-13:
We need some thin layer on top of [TLB](http://nxr.netbsd.org/xref/src-freebsd/sys/mips/mips/tlb.c). We'll have three types of entries:
* `wired` - non-removable pages that should always be mapped,
* `kernel` - pages placed in kseg2 and later kseg3 too,
* `user` - pages that belong to several address spaces mapped in useg.

Let's consider a few routines:
* `void tlb_dump_state()` - that one would be nice for obvious reasons,
* `void tlb_update(asid_t, vaddr_t va, pte_t pte)` - if an entry has been modified (e.g. protection bits) in *page table* or *page directory* we would like to modify it in TLB as well,
* `void tlb_insert_wired(uint32_t i, vaddr_t va, pte_t pte0, pte_t pte1)` - to insert non-removable entries.

Following procedure have to be executed only when a mapping (between *vaddr_t* and *paddr_t*) has been removed:
* `void tlb_invalidate_address(asid_t, vaddr_t va)`
* `void tlb_invalidate_range(asid_t, vaddr_t start, vaddr_t end)`

... or an address space has been removed:
* `void tlb_invalidate_all()`
* `void tlb_invalidate_all_user(asid_t)`

It is unclear to me how to recover from TLB information than within given time quantum a page was referenced or modified, which is essential for swapping mechanism.

coodie, 2016-05-13:
Some of these functions are either on pmap.c branch, or in mips/cpu.h. As for TLB, I think possible implementation issues will appear when I'll work on vm_map.

coodie, 2016-10-06:
Since introduction of vm_map, can we consider this task already closed? I know that both of these need maintenance, but in general problem is well understood and the code which we have actually works.

cahirwpz, 2016-10-07:
Well... in fact it requires further development ;-) IMO goals for this task were met, so I guess it's right time to close it.

closed 2016-10-07 by cahirwpz