cahirwpz/mimiker #52: Segment start and end address are thrashed while using vm_phys_reserve.

cahirwpz, 2016-05-10:
I think I encountered a bug in your code. I modified the code to show how to reproduce the bug:

https://github.com/cahirwpz/mimiker/blob/master/main.c#L75

coodie, 2016-05-10:
For some reason value of seg->start is changed here (vm_phys.c:124):

`TAILQ_INSERT_HEAD(PG_FREEQ(seg, page->order), page, freeq);`

For some reason changing this to:
`TAILQ_INSERT_TAIL(PG_FREEQ(seg, page->order), page, freeq);`

Invokes TLB refill handler, which shouldn't happen.

This is quite weird. I'll keep on investigating.

coodie, 2016-05-10:
vm_phys_reserve should receive only aligned addresses. Aligning addresses fixes this bug, but I guess there is more serious bug which is yet to find.

commit 21a6ca1e (2016-05-12): This fixes #52
closed 2016-05-13 by coodie