cahirwpz/mimiker #53: Clever (?) trick to make physical and virtual address distinct types

cahirwpz, 2016-05-13:
I'd like to avoid situation where we mix up physical and virtual without any control, here's first idea how to handle it:

```
typedef union {
  void *ptr;
  intptr_t addr;
} paddr_t;

typedef union {
  void *ptr;
  intptr_t addr;
} vaddr_t;

/* by explicit casting one can convert underlying type to paddr_t */
paddr_t foo1(void *addr) {
  return (paddr_t)addr;
}

/* by virtue of union's field access one can get value of underlying type */
void *foo2(paddr_t pa) {
  return pa.ptr;
}

/* and this is the way how one can convert between virtual and physical address */
paddr_t foo3(vaddr_t va) {
  return (paddr_t)va.ptr;
}
```

cahirwpz, 2016-05-13:
Give my your opinion please. Could you try it out and possibly convert all sources to use this idea?

coodie, 2016-05-13:
This is actually very nice idea. At the moment I'm probably the only person which can handle mess with vm_addr and p_addr properly. I like use of union, this will avoid use of casting between pointer and non-pointer type. Keep in mind that, this will however only solve type problems at compile time. For example your `0b109a4f` commit would probably type (line 68), but you'd need to know  that you should subtract proper offset from given address which wouldn't be discovered by type system, but (possibly) by kernel panics and asserts in proper places. This would also make our code a little bit ugly. Every place which would look like: `foo->bar = paddr; ` would turn into `foo->bar = paddr.addr`.
Do you know if there is any other way to avoid explicit casts in function calls? For example by adding some function attributes.

cahirwpz, 2016-05-13:
If you consider passing arguments to function, then transparent unions may be helpful IIUC: http://toves.freeshell.org/xueg/

cahirwpz, 2016-05-13:
With transparent unions following example works:

```
typedef union __attribute__((transparent_union)) {
  void *ptr;
  intptr_t addr;
} paddr_t;

typedef union __attribute__((transparent_union)) {
  void *ptr;
  intptr_t addr;
} vaddr_t;

paddr_t foo3(vaddr_t va);

paddr_t foo4(void *addr) {
  return foo3(addr);
}
```

coodie, 2016-05-13:
That is nice feature to have if we have a lot of unions in our code. This might make change a little bit less tedious.

What I meant is something like this:
```
typedef uint32_ptr paddr_t; 
typedef uint32_ptr vaddr_t;

void foo(paddr_t); 
```

And then code like:

```
vaddr_t x = ...;
foo(x); //implicit typecast from vaddr_t to paddr_t, doesn't compile when vaddr_t and paddr_t are union types
```

Would emit compile time error or at least warning about incompatible types. This might avoid some mistakes, and should be enough for potential user of subsystem, because developer of subsystem (if he is competent) knows what's going on.

cahirwpz, 2016-05-13:
AFAIK it's impossible to do so, but I'm not omniscient. This can be partially emulated by union trick I presented.

cahirwpz, 2016-07-04:
Meh :P

closed 2016-07-04 by cahirwpz