cahirwpz/mimiker #86: Failing testcase for switching vm_maps

rafalcieslak, 2016-10-28:
I have probably found a test case that proves we are not switching vm spaces correctly:
```

void vm_map_switching_demo(){
  vm_map_t *umap1 = get_active_vm_map(PMAP_USER);
  vm_map_t *umap2 = get_active_vm_map(PMAP_USER);
  
  vm_addr_t start = 0x1001000;
  vm_addr_t end = 0x1001000 + 2 * PAGESIZE;
  
  vm_map_entry_t *data1 =
    vm_map_add_entry(umap1, start, end, VM_PROT_READ|VM_PROT_WRITE);
  vm_map_entry_t *data2 =
    vm_map_add_entry(umap2, start, end, VM_PROT_READ|VM_PROT_WRITE);
  
  data1->object = default_pager->pgr_alloc();
  data2->object = default_pager->pgr_alloc();

  volatile char* p = (char*)(start + 0x10);
  
  set_active_vm_map(umap1);
  *p = 0x10;
  set_active_vm_map(umap2);
  *p = 0x20;
  set_active_vm_map(umap1);
  assert(*p == 0x10);
  set_active_vm_map(umap2);
  assert(*p == 0x20);
  
  log("Test 2 passed.");
}
```
When I add it to `./vm_map.c`, I get:
```
[panic] vm_map.c:66 Assertion '*p == 0x10' failed!
```
which suggests the second switch to `umap` did not happen. I suppose the reason for this may be that the page corresponding to `*p` is still in TLB. We're not yet flushing TLB on vm map switch, are we? The fix may be not trivial, as simply adding `tlb_invalidate_all` to `set_active_vm_map` starts a cascade of recursive page faults.

@coodie could you take a look at this? Is my test case even correct?

coodie, 2016-10-28:
```  
vm_map_t *umap1 = get_active_vm_map(PMAP_USER);
vm_map_t *umap2 = get_active_vm_map(PMAP_USER);
```

Are you sure this is correct? umap1 and umap2 should be different.
I think this is what you want to do:

```  
vm_map_t *umap1 = vm_map_new(USER_VM_MAP, 7);
vm_map_t *umap2 = vm_map_new(USER_VM_MAP, 42);
```

coodie, 2016-10-28:
Note that when we don't initialize user vm_map on kernel startup, so in my case this test made:
```
  vm_map_entry_t *data1 = vm_map_add_entry(umap1, start, end, VM_PROT_READ|VM_PROT_WRITE); //uma1 is null
```

nullptr exception, however my 'fix' works.

rafalcieslak, 2016-10-28:
Ah yes, you are correct. Silly me.

But since we're at it, I don't really like the fact that I need to specify a unique number while creating a new `vm_map`. Shouldn't this be done somewhere deeper in the interface? Especially with ASID's being a machine-specific feature.

coodie, 2016-10-28:
I realised this while working on vm_space switch for user threads. In my opinion it's best to hide asid in pmap, in very distant future we will need to make some asid allocator, but for now simple increment until we run out of asids should be enough.

rafalcieslak, 2016-11-04:
Closing, no longer relevant.

closed 2016-11-04 by rafalcieslak