cahirwpz/mimiker #143: FPU state not saved across user-mode context switches

cahirwpz, 2016-11-19:
Because FPU is not used inside kernel, its state is not saved during `user_exc_enter`. However it must get saved and restored properly on context switches as user-space program are allowed to perform floating point computations.

In FreeBSD routines that handle FPU context are named [MipsSwitchFPState](http://bxr.su/FreeBSD/sys/mips/mips/swtch.S#403) and [MipsSaveCurFPState](http://bxr.su/FreeBSD/sys/mips/mips/swtch.S#524). However it is not clear when exactly FPU context gets switched.

rafalcieslak, 2017-03-25:
> However it is not clear when exactly FPU context gets switched.

FreeBSD uses the usual optimisation of not switching FPU context when user programs are switched. Instead, the coprocessor 1 is disabled entirely, and when the user program tries to perform any FPU operation, a "coprocessor unusable" exception is raised. The handler checks if the referenced coprocessor is indeed coprocessor 1, and only then it restores FPU registers state and re-enables coprocessor 1 afterwards ([reference](http://bxr.su/FreeBSD/sys/mips/mips/trap.c#999)).

We can implement the same trick. At the moment we only have a single user-space program running at any time, I suppose we should postpone this issue until we have some kind of `fork` in place.

dingorth, 2018-04-07:
Here is a fix for this bug: #409 with #410.

commit 2c475bc1 (2018-05-01): Save & restore FPU context in a lazy way (#409)

* Fixes #143
closed 2018-05-01 by cahirwpz