cahirwpz/mimiker #145: Simple VFS interface and ramdisk filesystem

cahirwpz, 2016-11-19:
Simple interface should be devised for file operations. It'll be a subset of [fileops](http://bxr.su/FreeBSD/sys/sys/file.h#119) functions, namely: `open`, `read`, `stat`, `seek` and `close`. Each file is assigned [vnode](http://bxr.su/FreeBSD/sys/sys/vnode.h#97) structure, which is bound to filesystem's object representation.

After necessary abstractions are introduced, one can implement very simple filesystem. It's going to be a ramdisk in [cpio(5)](https://www.freebsd.org/cgi/man.cgi?query=cpio&amp;sektion=5) format visible under `/initrd` location.

rafalcieslak, 2016-11-20:
I've been thinking a lot about a reasonable way of splitting this task into smaller gradual changes. I believe the following scenario would be a good start:

1. First we would implement file/fileops interface for UART input/output. This would let us discuss and prepare the framework for working with files (global open files list, descriptor handling, `filedesc` structure etc.).
2. Then we might implement basic file-based operations that do not require an underlying filesystem, like `pipe` (simplified) or `dup2`. These would let us test the interface and prepare for working with real files. It also makes sure we keep the file interface generic. (Also, `pipe` might be interesting to use with a `clone`!)
3. Finally we would implement actual filesystem, including ramdisk and vnodes.

Does this sound like a good plan, or would you rather do it bottom-up (filesystem first, high-level function later)?

cahirwpz, 2016-11-22:
Short answer – let's get working ramdisk ASAP. We'll then replace embedded ELFs and provide working stubs for some system calls.

coodie, 2016-11-22:
I'd like to have a little rest from mutexes, can I instead work for a while on ramdisk implementation? This sound interesting to me as I am not very familiar with this.

cahirwpz, 2016-11-22:
Sure, just make sure @rafalcieslak doesn't work on the same problem.

@rafalcieslak Let's try to design `filedesc` interface for `/dev/null`.

rafalcieslak, 2016-11-22:
We've initially agreed that @coodie will investigate ramdisk creation, loading and access, while I will learn about file interfaces, including descriptor management. This way we'll start working from opposite sides and hopefully we'll meet in the middle.

cahirwpz, 2017-03-24:
VFS interface has been added. We're waiting for #184 to be finished as well.

rafalcieslak, 2017-04-19:
Merging #184 finalized basic ramdisk filesystem support. Additional features are tracked in separate issues.

closed 2017-04-19 by rafalcieslak