cahirwpz/mimiker #156: Interrupts are enabled before threads are initialized

rafalcieslak, 2016-11-29:
Having heard that @dyniec has problems with QEMU, I spent this day investigating what happens when running mimiker on QEMU, and I've found some pretty interesting stuff. Obviously, as it always turns out, it is by mere luck that the system runs okay-ish on OVPsim.

So the first problem is something we were already aware of: QEMU always jumps to `ebase + 0x200` whenever any interrupt arrives. Vectored interrupts are incorrectly supported in QEMU (see: [helper.c:674](https://github.com/qemu/qemu/blob/master/target-mips/helper.c#L674)), and we can either fix that ourself (shouldn't be difficult, but I doubt we want to maintain our qemu fork), or prepare two variants of our exception handlers - one for QEMU, using a single interrupt address `ebase + 0x200`.

As for now, this is how we process such interrupt:
```
irq0:
1:      j       b1
        nop
```
which causes an endless loop whenever QEMU emits an interrupt. So in order to *temporarily* fix this, I changed it into:
```
irq0:
1:      j       irq7
        nop
```
since the clock interrupt is the only interrupt we currently use anyway.

But that led to even more confusing problems. To explain their cause, please have a look at how we currently do `kernel_boot`, taking note on when are interrupts enabled:

```c
  cpu_init();
  pcpu_init();
  pci_init();
  pm_init();
  intr_init();      // <------- This calls enable_interrupts
  callout_init();
  tlb_init();
  pmap_init();
  vm_object_init();
  vm_map_init();
  sched_init();
  sleepq_init();
  mips_clock_init();  // <------ This enables clock interrupt, also disables and re-enables interrupts
  kprintf("[startup] subsystems initialized\n"); // <----- This takes, relatively, a long time
  thread_init((void (*)())main, 2, argc, argv);
```

Thus it is totally possible for a clock interrupt to arrive while we are printing "subsystems initialized", and it happens quite often when running mimiker under QEMU. Suppose that happens. To process an interrupt, we start by calling `kern_exc_enter`, which stores some data onto the stack, but also accesses current thread data (`LOAD_PCPU` and `PCPU_CURTHREAD`). However, `_pcpu_data.currthread` was not set yet, because `thread_init` wasn't called yet! This, naturally, lead to immediate crash by dereferencing a null pointer.

Thus we need to either:
1. Enable interrupts only after the first thread has started. This would require us to adopt a policy that functions called in `kernel_boot` may not enable interrupts, and may not enter critical sections. Currently only `mips_clock_init()` does that. However, `thread_init` uses generic context switching functions to run the first thread, and these use (among others) `vm_map_activate` and `pmap_activate`, which use a critical section. Therefore we would also need a clever way of starting the first thread that does not re-enable interrupts in the process, and we would enable them in `main` (or a `pre-main`).
2. Set `_pcpu_data.currthread` much earlier, so that interrupts may be enabled during the startup process. I'm not sure how to do that. `thread_init` does not set `currthread` directly, it happens when `ctx_boot` runs `ctx_resume` - which we can't call earlier. Choosing this option would require us to reorganize the `thread_init` process, possibly complicating it a bit.

I don't know what would be the better choice, or if there are other options (I suppose that removing references to `currthead` in `kern_exc_enter` is impossible).

---

A side-problem I noticed (I tested QEMU with the first proposed fix applied in a brute-force way) is that QEMU triggers interrupts much more frequently than we can handle, so a new interrupt arrives before we are done with the previous one. This may be caused by either:

a) QEMU incorrectly triggerring too frequent clock interrupts
b) QEMU triggering interrupts we have not enabled
c) QEMU providing incorrect values for clock interrupts, causing us to spend infinite amount of time when processing clock interrupts.

I'll investigate this side-problem further once we get interrupts to work correctly.

rafalcieslak, 2016-11-29:
Oh, it seems that [ImgTech worked on QEMU interrupts](https://github.com/qemu/qemu/commit/da52a4dfcc4864fd2260ec4eab331f75b1f0240b) not long ago. I'll see whether this change fixes anything for us.

cahirwpz, 2016-11-30:
It's not a secret that interrupts handling is neglected to large extent, hopefully @dyniec will take care of it. I tried various approaches to make use of vectored interrupts [recently](https://github.com/cahirwpz/mimiker/commit/2adbf0e60890c92d46c9cc679ad822dce550d611) and frankly... seems like single entry point makes the most sense.

The problem you're describing is more profound and already emerged when we tried to integrate sleepqueues. At the very beginning I assumed that kernel initialization environment is uniform, which is far from truth. By analysis of [mach_init](http://bxr.su/NetBSD/sys/arch/evbmips/malta/machdep.c#146) in NetBSD and [platform_start](http://bxr.su/FreeBSD/sys/mips/malta/malta_machdep.c#266) one can learn that many of `kernel_boot` steps should be moved to `platform_init`, and some assumptions about working environment for both procedures must be made.

I'm still in process of reading both files and making conclusions. The problem must be fixed ASAP.

PR #158 (2016-12-06): Split kernel startup into two phases
This PR fixes various issues with kernel startup phase. Firstly it splits previous `kernel_init` into `platform_init` and `kernel_init`. The former does various hardware dependant initialization steps and creates a thread that will execute `kernel_init` procedure. The latter does machine independent stuff, runs scheduler and never returns. Secondly makes PCI initialization use regular memory allocator. Thirdly it fixes some mistakes in sw & hw interrupts handling routines. 

The outcome of this effort is that tests run with QEMU, which might be of interest  [... cut]
closed 2016-12-06 by cahirwpz