cahirwpz/mimiker #159: Physical and virtual memory management subsystems overhaul

cahirwpz, 2016-12-06:
This issue was submitted purely for design and discussion purposes.

We want to address some of the deficiencies of `physmem`, `pmap` and `vm_map` subsystems including:
* lack of page sharing between virtual spaces
* emulation of `modified` and `referenced` bits for MIPS architecture
* moving page directory table to virtual memory
* ... and many others we're going to list here as work progresses

Corresponding NetBSD 9 manuals: [pmap(9)](http://netbsd.gw.com/cgi-bin/man-cgi?pmap+9+NetBSD-current), [uvm(9)](http://netbsd.gw.com/cgi-bin/man-cgi?uvm+9+NetBSD-current) and FreeBSD 11 manuals: [pmap(9)](https://www.freebsd.org/cgi/man.cgi?query=pmap&apropos=0&sektion=9&manpath=FreeBSD+11-current), [vm_map(9)](https://www.freebsd.org/cgi/man.cgi?query=vm_map&sektion=9&apropos=0&manpath=FreeBSD+11-current).

strzkrzysiek, 2016-12-09:
What was discussed so far is the basic management physical memory frames.
Right now functions from `physmem.h` work in terms of _vm_pages_. We agreed that it is wrong to keep inside this structure virtual addresses since a page (or maybe frame?) should be able to be mapped many times at once. The idea is to handle physical memory in terms of `pgframe_t` structure which represents a 4kB block.
```C
typedef struct pgframe {
  struct pgframe * link;
  pm_addr_t paddr;
  unsigned char flags;
} pgframe_t;
```
Page frames can be linked together in order to represent larger amount of memory.

Basic `physmem.h` set of functions should look like this then:
```C
pgframe_t * pm_alloc(size_t bytes);
void        pm_free(paddr_t start, size_t bytes);
void        pm_steal(paddr_t start, paddr_t end);
void        pm_boot_add_segment(paddr_t start, paddr_t end);
pgframe_t * paddr_to_pgframe(paddr_t pa);
```

Another thing is the location of _page directory table_. It can be moved from `kseg0` to `kseg2` and be placed just after all PTEs ([image](http://imgur.com/a/KD5t4)).

cahirwpz, 2019-02-13:
Several issues listed in here has already been solved. Some of them will return in separate PR's.

closed 2019-02-13 by cahirwpz