cahirwpz/mimiker #161: Is it okay to tlb_print in tlb_exception_handler?

rafalcieslak, 2016-12-06:
Currently, in `void tlb_exception_handler(exc_frame_t *frame)` we do:

```c
log("%s at $%08x, caused by reference to $%08lx!", exceptions[code],
    frame->pc, vaddr);
tlb_print();
```

On numerous occasions already this has caused me problems with `tlb_print` looping infinitely (recursively?). It seems as if a tlb exception was triggered while printing. Could this be the case? From my experience, the address referenced (`vaddr`) in such troublesome case is usually in range `0xc*******`. Isn't this where we keep our page table?

I've ignored this problem until today, when I found out that removing `tlb_print` from `tlb_exception_handler` apparently fixes the issues I've experienced - the tlb gets refilled, and any further data access appears correct.

So should we strictly abstain from printing tlb status in the exception handler, or is there something I'm missing?

cahirwpz, 2016-12-07:
Can you explain how to reproduce this bug ?

rafalcieslak, 2016-12-07:
I can't exactly pinpoint the problem, so the test case I can provide is not simple. You need to `malloc` 6MiB (as a userspace program). For some reason it has to be exactly 6MiB, many other sizes I've tested do not trigger as many bugs as 6MiB - in fact, I've been using this test for weeks, and it already helped me discover numerous bugs.

So to reproduce this issue, use this userspace program (or add this to the beginning of an existing one):

```c
int main(){
  int *ptr = malloc(6 * 1024 * 1024);
  free(ptr);
}
```
`malloc` uses `sbrk` and then writes some data into the allocated memory.

rafalcieslak, 2016-12-11:
With #166 merged, I hope this is fixed. If I find another way to trigger problems with TLB, I'll reopen or create a new issue.

closed 2016-12-11 by rafalcieslak