cahirwpz/mimiker #189: Kernel logging facilities for debugging

cahirwpz, 2017-01-19:
The interface proposal is in [klog](https://github.com/cahirwpz/mimiker/blob/klog/include/klog.h) branch. It's modeled after [ktr(9)](https://www.freebsd.org/cgi/man.cgi?query=ktr&sektion=9) facility from FreeBSD.

The idea is too keep circular array of `klog_entry_t`. Appending to kernel log requires almost to synchronization – just incrementing head pointer will do. The messages will not be printed to the console, unless `klog_verbose` parameter is passed to the kernel. Thus such mechanism should have little impact on running kernel.

There're many sources of kernel log messages (for instance: `KL_SCHED`, `KL_VM`, etc.). A user can specify through  `klog_mask` kernel parameter which events they want to keep in log buffer. One should be able to print the contents of the buffer by invoking **gdb** extension script.

rafalcieslak, 2017-04-28:
#239, which was recently merged, introduced basic interface for message logging. However, it's not yet widely used in the kernel.

cahirwpz, 2017-05-18:
From now on almost `klog` is used among most kernel files.

closed 2017-05-18 by cahirwpz