cahirwpz/mimiker #213: callout improvements

cahirwpz, 2017-03-24:
Callouts must be made thread-safe! We have to consider where to execute timeout callbacks. Probably high-priority worker thread is the best place.

Additionally we need following changes to the interface:

```C
void callout_schedule(callout_t *c, unsigned time);
void callout_set_func(callout_t *c, timeout_t fn, void *arg);
void callout_reset(callout_t *c, unsigned time, timeout_t fn, void *arg);
```

Please note that `callout_setup*` functions are replaced by `callout_schedule` and `callout_set_func`. Callout functions use `time` measured in milliseconds. Functions accept time relative to the time kept by `hardclock`.

For exact meaning of each function please look at [callout(9)](http://netbsd.gw.com/cgi-bin/man-cgi?callout+9+NetBSD-current).

**IMPORTANT** I do not expect timeout callbacks to be called in order!

rafalcieslak, 2017-04-28:
Callout implementation improvements are currently blocked by #216 (Worker thread).

cahirwpz, 2018-07-24:
Seems like callouts were greatly improved in c64a9ac72726999483b499b27e3f38d8ec14af10, 5ce803bc4dd0ae064b936dc8b449eec7f076d7c9. IMO at the moment callout API is in good condition.

closed 2018-07-24 by cahirwpz