cahirwpz/mimiker #227: `thread_join` state check and sleep are not done atomically

rafalcieslak, 2017-03-26:
Currently `thread_join` uses a sleep queue:

```
  if (otd->td_state == TDS_INACTIVE)
    return;

  sleepq_wait(&otd->td_exitcode, "Joining threads");
```

It is unlikely that the joined thread `odt` would exit after we've checked its state, but before we `wait` for it to exit - but that is possible, and that's exactly what happened [in this Travis build](https://travis-ci.org/cahirwpz/mimiker/builds/215327055).

Normally I would use a conditional variable in such case... Is there some other trick we might use?

(Wrapping this into a critical section won't work because the sleep would never terminate. Also: a timeout for `sleepq_wait` would not be an elegant solution for this issue.)

rafalcieslak, 2017-03-27:
I've noticed both `mutex` and `rwlock` successfully call `sleepq_wait` from within a critical section. I'm not sure why does it work, I expected this would permanently disable interrupts and thus stop the scheduler. Anyway, this should probably solve this issue.

cahirwpz, 2017-03-27:
I'll explain it tomorrow. Please notice that context switch saves and restores `SR_IE` flag correctly.

rafalcieslak, 2017-04-19:
I believe a combination of #229 and #265 fixed this bug.

closed 2017-04-19 by rafalcieslak