cahirwpz/mimiker #241: `critical_exit` should switch to another thread if indirectly requested

cahirwpz, 2017-04-06:
Intuitively [critical_leave](http://mimiker.ii.uni.wroc.pl/source/xref/mimiker/sys/sync.c#12) should switch to another thread if, by execution of critical section, a thread was added to run queue and the thread has a higher priority or currently running thread's time slice was exhausted. But we don't do it.

By looking at FreeBSD's [critical_exit](http://bxr.su/FreeBSD/sys/kern/kern_switch.c#197) it's obvious that they handle it correctly by utilizing `td_owepreempt` field in `thread`'s structure.

```C
/* k*- only accessed by curthread and from an interrupt */
volatile u_char td_owepreempt;  /* (k*) Preempt on last critical_exit */
```

The flag is set by [maybe_preempt](http://bxr.su/FreeBSD/sys/kern/sched_4bsd.c#maybe_preempt) which is called by [sched_add](http://bxr.su/FreeBSD/sys/kern/sched_4bsd.c#1396). The latter is called in many places, but one is of particular interest i.e. [turnstile_unpend](http://bxr.su/FreeBSD/sys/kern/subr_turnstile.c#939), which wakes up threads that were suspended on a turnstile.

melouber, 2018-04-25:
In Mimiker we already set `TDF_NEEDSWITCH` when time slice is exhausted or when some more important thread is added to run queue. I'd say that solution to this issue is adding this code:
```
if (td->td_pdnest == 0 && td->td_flags & TDF_NEEDSWITCH) {
  WITH_SPINLOCK(td->td_spin) {
    td->td_state = TDS_READY;
    sched_switch();
  }
}
```
at the end of `preempt_enable` function. What do you think?

cahirwpz, 2018-04-27:
Looks reasonable to me. I'll accept the change if you create PR with it.

commit 47dffebc (2018-04-29): `preempt_enable` switches to another thread if time slice was exhausted

* Fixes #241
closed 2018-04-29 by cahirwpz