cahirwpz/mimiker #245: Inter-process communication with pipes

cahirwpz, 2017-04-07:
Pipes are unidirectional method of inter-process communication described in [pipe(2)](http://netbsd.gw.com/cgi-bin/man-cgi?pipe+2+NetBSD-current).

Pipes are another [type](http://bxr.su/NetBSD/sys/sys/file.h#161) of file with corresponding [fileops](http://bxr.su/NetBSD/sys/kern/sys_pipe.c#114) structure. [pipe](http://bxr.su/NetBSD/sys/sys/pipe.h#102) structure stores file related metadata, synchronization primitives and [pipebuf](http://bxr.su/NetBSD/sys/sys/pipe.h#63) cyclic buffer. [uiomove](http://netbsd.gw.com/cgi-bin/man-cgi?uiomove+9+NetBSD-current) facility is used to copy data between the buffer (placed in kernel virtual address-space) and source and destination processes.

cahirwpz, 2017-06-07:
@MoloToha Any progress on the task? It's a pretty exciting feature so we'd like to have it sooner rather than later.

MoloToha, 2017-06-07:
@cahirwpz I'll try my best to finish this until the next meeting. Last week I was busy with other subjects so I didn't make any progress, although the task is close to be done (you can see it on pipe branch).

cahirwpz, 2017-06-07:
Could you prepare a PR with [WIP](https://github.com/cahirwpz/mimiker/wiki/Github-issue-PR-labels) label so we could have a look at it?

cahirwpz, 2018-07-24:
Implemented in c9068b7b69bf71f5f4fa0563e6a56f4880f13389

closed 2018-07-24 by cahirwpz