cahirwpz/mimiker #247: File descriptor table improvements

cahirwpz, 2017-04-07:
For simplicity file descriptor table is limited to `MAXFILE` entries and the last usable descriptor number is `NDFILE`. At some point we will be interested to have more than 20 resources opened by a single process. The task is to redesign the data structure for file descriptor table [fdtab](http://mimiker.ii.uni.wroc.pl/source/xref/mimiker/include/filedesc.h#16) so it can grow and shrink.

MoloToha, 2017-04-17:
I would suggest to create a separate structure, it's sketch looks like that 
```
typedef struct fdt_chunk {
  TAILQ_ENTRY(fdt_chunk) fdtc_chunk_list;
  file_t *fdtc_files[NDFILE];
  bitstr_t fdtc_map[bitstr_size(NDFILE)];
  int fdtc_id;
} fdt_chunk_t;
```
Such chunk contains an array of `NDFILE` pointers to file objects. A file in each chunk has `fdtc_id * NDFILE + offset` file descriptor (where `fdtc[offset]` is a pointer to file), and we can find a file with descriptor `fd` by finding a chunk with id `fd / NDFILE` and getting a pointer `fdtc_files[fd % NDFILE]`. With this approach, we substitute a fixed size array with a flexible tail queue `TAILQ_HEAD(, fdt_chunk_t) fdt_list` to which we can easily attach another chunk (it's id will be the id of last chunk in the queue plus 1), if needed, and do a cleanup of empty chunks, all at cost of performance decrease up to 52 times (if we set the limit of opened files to 1024 and `NDFILE` to 20).

rafalcieslak, 2017-04-17:
Isn't this an overly complicated solution for a simple problem? What FreeBSD does is just allocating a completely new, contiguous array for file pointers whenever a process runs out of free descriptors. See [this comment](http://bxr.su/FreeBSD/sys/kern/kern_descrip.c#134), and [this function](http://bxr.su/FreeBSD/sys/kern/kern_descrip.c#1503). Our file descriptor table is heavily based on FreeBSD's [struct filedesc](http://bxr.su/FreeBSD/sys/sys/filedesc.h#68), so you will see many similarities to our implementation.

Remember that **all** system calls that reference anything file-like require a lookup in the descriptor table. Traversing 52 entries on a list on each `read`/`write`/`seek`/`stat`/... would be a lot of work (and if that's a dynamically extended list, we may even encounter page faults while traversing it)! Okay, copying data to a new, larger array does cost some time too, but it only has to be done when a new file descriptor is referenced (`open`,`dup`,etc.) - which is, relatively, very rare. And if the table grows in size exponentially (e.g. we double its size every time it has to grow), then the amortized complexity gets comfortably low.

Also, I suppose using a single, contiguous array for file pointers results in much simpler implementation than splitting the array into chunks.

MoloToha, 2017-04-17:
I agree with you, this solution looks much better. I think I can implement it in the meantime, while another pull request is waiting for review

cahirwpz, 2017-04-25:
Implemented with #273

closed 2017-04-25 by cahirwpz