cahirwpz/mimiker #249: Need more drivers!

cahirwpz, 2017-04-07:
**Work In Progress!**

Buses:
* processor (?)
* CBUS
* PCI
* ISA
* IDE

Drivers:
* CPU-PCI bridge
* PCI-ISA bridge
* 16C550 UART
* RTC
* PS/2 keyboard and mouse

rafalcieslak, 2017-04-13:
For the record, I am working on a prototype device/driver/bus interface with PCI support stub. The major breakthrough was realizing the the LDD3 book (https://lwn.net/Kernel/LDD3/) is much more concise and clear than FreeBSD books and docs - thus my implementation may be slightly skewed towards Linux style, but that shouldn't matter, given how generic these interfaces are.

rafalcieslak, 2017-04-14:
Some further remarks:

A CBUS bus interface is irrelevant. This kind of bus is **only** present on the malta board, and nowhere else. No operating system implements an abstraction over the CBUS, they manually prepare a direct device for CBUS UART in malta-specific code ([example1](http://lxr.free-electrons.com/source/arch/mips/mti-malta/malta-platform.c), [example2](http://bxr.su/FreeBSD/sys/mips/malta/uart_bus_maltausart.c)). Given how specific it is, I see no value in implementing it as a particular bus type.

It is not possible to enumerate over available ISA devices. As ISA register addresses are fixed and all machines use the same set of addresses for a particular piece of hardware, the abstraction an ISA bus provides is very thin ([Linux example](http://lxr.free-electrons.com/source/drivers/base/isa.c)). In fact, the ISA interface in Linux is pretty [almost unused](http://lxr.free-electrons.com/ident?i=isa_register_driver). In many cases ISA devices are implemented as platform-specific, always available devices. Again, this removes a layer of abstraction.

I am unable to find any traces of support for `FDC37M817` in neither Linux, FreeBSD nor NetBSD. Usually some features provided by this chip (16550) are implemented as platform specific devices. Others are only initialized and never referenced.

This leaves me in a weird situation where it is not possible imitate other's solutions when implementing the device/bus abstractions, because most devices we need (UARTs, buses, keyboards) are supported in other operating systems bypassing a generic interface.