cahirwpz/mimiker #254: Missing pmap_delete() on vm_map_delete()

goniz, 2017-04-09:
Hi

I've observed some unwanted behavior when running this pseudo code:
`
pid_t pid1 = fork_exec("/bin/ls");
waitpid(pid1);

pid_t pid2 = fork_exec("/bin/ls");
waitpid(pid2);
`
after waitpid "collects" the first process, the second one fails to exec because it gets the same vm_map/pmap struct addresses that were in an undefined state..

we need to call pmap_delete() in vm_map_delete() to balance the pmap_new() that gets called in vm_map_new().

I'm sorry I do not have the time to submit a patch, but I thought I'd better let you know.

goniz, 2017-04-09:
Another observation: we need to flush the tlb before we reuse the same asid

rafalcieslak, 2017-04-28:
These issues will be addressed by the major memory management rewrite that has been in progress since few months already. You can track *some* initial progress in this PR: #190. Unfortunately, these fixes are not of our high priority at the moment.

cahirwpz, 2017-07-17:
Fixed by #392 (hopefully)

closed 2017-07-17 by cahirwpz