cahirwpz/mimiker #287: `do_lseek` ignores `whence` argument

rafalcieslak, 2017-05-02:
The third argument to `lseek` is `whence`, which may be one of:

 - `SEEK_SET` - the file offset is set to `offset` bytes
 - `SEEK_CUR` - the file offset is set to its current location plus `offset` bytes
 - `SEEK_END` - the file offset is set to the size of the file plus `offset` bytes.

Currently our `do_lseek` ignores `whence` and always works as `SEEK_SET` (it was fine for the programs we've tested until now), but we'll have to implement other flags as well. `SEEK_CUR` is trivial to implement, `SEEK_END` *probably* requires the underlying file to support `getattrs`.

coodie, 2017-05-02:
I have noticed that already. I plan to work on mimiker upcoming weekend and hope to provide this feature.

cahirwpz, 2017-05-18:
@coodie What is the status of this task? Paweł Dyniec (@dyniec) expressed an interest in fixing the bug.

coodie, 2017-05-18:
@cahirwpz
That's easy to fix, and the main difficulty is going to be integration of user space library with our kernel, that's gonig to be quite tedious and if @dyniec really wants to fix it, let him do it (for extra points!).

cahirwpz, 2017-06-05:
The Open Group Base Specifications Issue 7: [lseek](http://pubs.opengroup.org/onlinepubs/9699919799/functions/lseek.html)

cahirwpz, 2017-06-28:
Implemented with #358

closed 2017-06-28 by cahirwpz