cahirwpz/mimiker #293: Identyfing the quality of OVPsim device emulation

rafalcieslak, 2017-05-05:
We've recently run into problems with VGA and PS/2 device implementation in OPVsim. Tuning our drivers so that they are compatible with incomplete OVPsim support is a waste of time and is not educational. Before we decide how to adjust our preference of simulators, we need to gather evidence on the poor quality of OVPsim's devices. In particular, it is important to investigate how likely it is we'll run into major problems with other devices that we consider crucial, e.g. the IDE controller and a hard drive.

rafalcieslak, 2017-05-06:
About VGA support:
 - The only VGA device available for OVPsim is the CirrusLogic GD 4556.
 - Probably the most painful missing feature is MMIO via BAR2 support, it's explicitly marked as unimplemented.

About PS/2 support:
 - Mouse is unsupported, this is even stated in the Malta platform documentation.
 - Scancode table selection is not implemented. The command is ignored. Always uses scancode table 2.
 - Some keys don't generate MAKE codes, only BREAK.
 - Some keys (e.g. cursor direction, keypad, control keys) use incorrect scancodes.
 - I've investigated the source code closely. It's clear that the sources are based on QEMU's. You can find following comments in sources `//qemu_system_reset_request()`. Some important features are replaced with a comment like `/* ignore that - I don't know what is its use */`.
    - I've tried to discover the exact QEMU version these sources are based on. It turned out to be a difficult task, because downloads for QEMU < 0.10 are no longer available anywhere on the internet. And it is indeed the 0.10 version that most closely matches OVP sources. For example, compare QEMU 0.10.0 `hw/ps2.c` and OVPsim `ImperasLib/source/ovpworld.org/modelSupport/ps2/1.0/pse/ps2.c.h`. These files are almost identical, except that OVPsim has added more diagnostic output, and removed almost all comments that mention qemu. The fact that qemu's version has some extra logic in code and checks for additional conditions makes me believe that OVPsim's files are actually based on an even older QEMU (for some reason I believe it is 0.8.2). Note that we're talking about QEMU releases back from 2006-2007.
    - These aren't the only files borrowed from QEMU. In fact, probably all files providing sources for the Malta platform I've read have striking resemblances to their counterpart in a very old QEMU. This includes PS/2 keyboard, PS/2 controller, IDE controller, PCI bus emulation, CirrusLogic VGA. Only core logic files are taken from QEMU, the peripheral interface description (so parts which link the device with OVPsim plugin infrastructure) appear to be written from scratch (but these seem to be barely interface adapters).

About IDE controller:
 - I find it difficult to examine potential problems without actually trying to implement the driver and checking what doesn't work. However, given the `ImperasLib/source/intel.ovpworld.org/peripheral/PciIDE/1.0/pse/ide.c` is, again, a direct adaptation of QEMU sources, I expect it may contain problems of significance similar to the devices described above.

cahirwpz, 2017-05-31:
After discussing the topic we've decided to stop using OVPsim altogether. Its device emulation is in bad shape, to the point it's not possible to implement device driver according to chipset datasheets.

closed 2017-05-31 by cahirwpz