cahirwpz/mimiker #308: Ramdisk test displays binary junk

rafalcieslak, 2017-05-13:
Currently the `ramdisk` test passes, but the output it produces is incorrect. Example:

```
Running test ramdisk.
file /usr/include/sys/errno.h:
/* errno is not a global variable, because that would make using it
   non-reentrant.  Instead, its address is returned by the function
   __errno.  */

#ifndef _SYS_ERRNO_H_
#ifdef __cplusplus
extern "C" {
#endif
#define _SYS_ERRNO_H_

#include <sys/reent.h>

#ifndef _REENT_ONLY
#define errno (*__errno())
extern int *__errno _PARAMS ((void));
#endif

/* Please don't use these variables directly.
   Use strerror instead. */
extern __IMPORT _CONST char * _CONST _sys_errlist[];
extern __IMPORT intp]^�
```

Similar junk is visible at the end of all files `ramdisk` test dumps. The output buffer is previously initialized with `\0`s, so apparently these junk bytes are copied in by `VOP_READ`.

dyniec, 2017-05-17:
The `t` letter is 500th byte in `/usr/include/sys/errno.h` file. The buffer in `test_ramdisk` isn't null terminated.
```c
char buffer[500];
  memset(buffer, '\0', sizeof(buffer));

  uio_t uio;
uio = UIO_SINGLE_KERNEL(UIO_READ, 0, buffer, sizeof(buffer));
```

rafalcieslak, 2017-05-17:
It seems that you are correct. Passing `sizeof(buffer) - 1` to `UIO_SINGLE_KERNEL` fixes the problem for me. Do you want to prepare the fix, or should I do it?

dyniec, 2017-05-18:
I will prepare the fix.

cahirwpz, 2017-05-18:
Fixed with #325

closed 2017-05-18 by cahirwpz