cahirwpz/mimiker #310: exec_fd_test fail when compiled with -O0

coodie, 2017-05-13:
I have stumbled upon this bug when playing with getdirents.

There is very weird behaviour when executing exec_fd_test. The exact test which is failing is located in `fd_test.c` and it is `test_copy` function. What happens briefly is this:

```
  char *naughty_ptr2 = (char *)0x00001000;
  assert_write_fail(0, naughty_ptr2, 200, EFAULT);
```

When performing write on naughty_ptr kernel correctly notices that it is indeed trying to write into unmapped region of user space and then jumps to `tlb_exception_handler` where recovery occurs. Until this moment everything is correct, but when unrolling stack and trying to propagate error code up the stack encounters this routine:

```
LEAF(copyerr)
	REG_L	ra, CALLFRAME_RA(sp)
	PTR_ADDU sp, sp, CALLFRAME_SIZ
	j	ra
	li	v0, -EFAULT		# return error
END(copyerr)
```
Note the loading to `ra` register. When code is compiled with -Og the value of this register jumps to proper handling routine (not sure right now, but its probably memcpy), but when code is compiled with -O0 the `ra` contains value of `naugthy_ptr`. I tried to figure out what's going on here for approximately 2 hours, but couldn't do it. I think this is going to be very interesting and challenging to debug and someone might try to do it for fun therefore I decided to put this as issue.

rafalcieslak, 2017-05-13:
Kudos to you for investigating this! Looks like it's a very important issue. With `-Og` the compiler performs tail call optimization, which seems to have affected this procedure. I believe it is crucial for us to successfully compile/run with `-O0`, even if we normally use `-Og` - failure to do that usually indicates a major undefined behavior, or completely misunderstood implementation. Note that this problem possibly originates in a very sensitive part of the kernel, so if we don't fix this soon, it'll come to haunt us later.

Supposedly you may want to try `-O2` as well and see what happens... but here be dragons.

> this is going to be very interesting and challenging to debug

I know you *love* debugging this kind of bugs, but... I am also aware that you're worried that we don't normally get graded for fixing minor bugs, that spending many hours investigating a problem that may boil down to a bitesize fix just isn't worth it, and that it would be better for your score to focus on other tasks. Let me reassure you that I (and, presumably, @cahirwpz too) recognize the significance of this curious issue, and I am also aware of the effort that it may take to troubleshoot it properly. Therefore, please: do continue investigating this issue, if you want, and be informed that I, personally, would be *very disappointed* if you were to not receive any points for it.

cahirwpz, 2018-07-24:
I attempted to reproduce the bug with `-O0` enabled with no success. I guess we fixed the source of this bug some time ago.

closed 2018-07-24 by cahirwpz