cahirwpz/mimiker #311: Introduce long test flag for ktest

coodie, 2017-05-13:
With recent klog test introduction, many tests on travis on other branches started to fail due to random timeouts on klog test, therefore I decided to remove it from regression.

There is also a case of ramdisk_dump test, and as far as I can recall there were some problems with timeouts in synchronization.

I think we shouldn't give up on those 'long tests' because they are actually quite useful, but having a lot of long repeated tests in kernel is terrible in my opinion because this will make regression so much longer. My idea is to add flag `KTEST_LONG` for this type of tests. Tests marked as `KTEST_LONG` won't be run repeatedly, but instead they will be run just once with extended timeout.

cahirwpz, 2017-05-20:
Until we provide a reasonable solution please mark long running tests with `KTEST_FLAG_BROKEN`.

cahirwpz, 2017-05-25:
@czapiga For the record, I've just disabled long running malloc tests.