cahirwpz/mimiker #369: vnodeops and vfsops interface completeness

cahirwpz, 2017-06-06:
* [vnodeops(9)](https://www.freebsd.org/cgi/man.cgi?query=vnodeops&manpath=NetBSD)
* [vfsops(9)](https://www.freebsd.org/cgi/man.cgi?query=vfsops&manpath=NetBSD)

```c
/* Mimiker */ int VOP_LOOKUP(vnode_t *dv, const char *name, vnode_t **vp);
/* NetBSD7 */ int VOP_LOOKUP(vnode_t *dv, vnode_t **vp, struct componentname *cnp);

/* Mimiker */ int VOP_READDIR(vnode_t *dv, uio_t *uio, void *state);
/* NetBSD7 */ int VOP_READDIR(vnode_t *v, uio_t *uio, kauth_cred_t cred, int *eofflag,
                              off_t **cookies, int *ncookies);

/* Mimiker */ int VOP_OPEN(vnode_t *v, int mode, file_t *fp);
/* Mimiker */ int VOP_CLOSE(vnode_t *v, file_t *fp);
typedef int vnode_read_t(vnode_t *v, uio_t *uio);
typedef int vnode_write_t(vnode_t *v, uio_t *uio);
typedef int vnode_seek_t(vnode_t *v, off_t oldoff, off_t newoff, void *state);
typedef int vnode_getattr_t(vnode_t *v, vattr_t *va);
typedef int vnode_create_t(vnode_t *dv, const char *name, vnode_t **vp);
typedef int vnode_remove_t(vnode_t *dv, const char *name);
typedef int vnode_mkdir_t(vnode_t *v, const char *name, vnode_t **vp);
typedef int vnode_rmdir_t(vnode_t *v, const char *name);
```

* [componentname](http://bxr.su/NetBSD/sys/sys/namei.h#92)

cahirwpz, 2017-06-06:
Why in both `VOP_REMOVE` and `VOP_RMDIR` they have extra argument `v`? Seems like they can remove entries either by `v` or `cnp`. Is there any other reason for that?

```c
/* Mimiker */ int VOP_REMOVE(vnode_t *dv, const char *name);
/* NetBSD7 */ int VOP_REMOVE(vnode_t *dv, vnode_t *v, struct componentname *cnp);

/* Mimiker */ int VOP_RMDIR(vnode_t *v, const char *name);
/* NetBSD7 */ int VOP_RMDIR(vnode_t *dv, vnode_t *v, struct componentname *cnp);
```

cahirwpz, 2017-06-06:
Clearly we should have `va` argument to `VOP_CREATE` and `VOP_MKDIR`, because we need to know attributes like `uid`, `gid`, `mode` and so on to create a file entry.

```c
/* Mimiker */ int VOP_CREATE(vnode_t *dv, const char *name, vnode_t **vp);
/* NetBSD7 */ int VOP_CREATE(vnode_t *dv, vnode_t **vp,
                             struct componentname *cnp, vattr_t *va);
/* Mimiker */ int VOP_MKDIR(vnode_t *v, const char *name, vnode_t **vp);
/* NetBSD7 */ int VOP_MKDIR(vnode_t *dv, vnode_t **vp,
                            struct componentname *cnp, vattr_t *va);
```

We are missing `VOP_SYMLINK` in Mimiker...

```c
/* NetBSD7 */ int VOP_SYMLINK(vnode_t *dv, vnode_t **vp, struct componentname *cnp,
                               struct vattr *vap, char *target);
```

coodie, 2017-06-06:
One important thing you might notice is how are these functions to be actually used. 

Let's take `VOP_REMOVE` as an example:

Very important thing to notice is that there is no `remove` syscall or something that would resemble removing, there is just `unlink` which removes file, but it happens in a little bit more complicated way - it is more general, because some process might still read from it. 

```
     The unlink() system call removes the link named by	path from its direc-
     tory and decrements the link count	of the file which was referenced by
     the link.	If that	decrement reduces the link count of the	file to	zero,
     and no process has	the file open, then all	resources associated with the
     file are reclaimed.  If one or more process have the file open when the
     last link is removed, the link is removed,	but the	removal	of the file is
     delayed until all references to it	have been closed.  The path argument
     may not be	a directory.
```

It basically decreases "reference count" on a file and removes it if it was zero and nobody had it opened, otherwise it waits. That is however not important for us, because we decide when we want to call REMOVE.

According to unlink pseudocode for removal would look a little bit like this:

```c
do_unlink(const char *path)
{
  vnode_t *v = vfs_lookup(path); 
  if(vfs_dec_wait(v))
  {
    vop_remove(v);
  }
}
```

If you have extra name argument, then you have to split path into directory path and file name.

```c
do_unlink(const char *path)
{
  const char *dir = get_dir_from_path(path);
  const char *name = get_name_from_path(path);
  vnode_t *v = vfs_lookup(dir); 
  if(vfs_dec_wait(v))
  {
    vop_remove(v, name);
  }
}
```

This also boils down to capabilities of vfs_lookup and what it actually has to do. This might be somehow related to symbolic links, so in some cases name can be useful. I'm not sure what covers these.

Anyway. What I'm trying to say is that, you probably aren't going to come up with perfect interface at this moment, this will have to be changed in the future anyway, because it is impossible to spot all possible problems, note that for some reason NetBSD and FreeBSD interfaces differ, I'm 100% sure they changed in time.

I'd rather focus on some basic capabilities and try to see if this interface covered all I need. Note that In VERY long time (couple of years), we probably won't even persistent file system, and currently we only have need to abstract READ and WRITE (specifically for devfs files), so it is apparent to me that we need only one well thought and well implemented filesystem which will do all of operations you mentioned. Every devfs, tmpfs, initrd, procfs and whathever-in-ram-fs will have exactly the same tree-like structure and exactly the same operations, the only thing that will differ among these is the concept of what actually file is.

We have slightly different approach to design. I like to think for a little while and then create prototype and see what are potential problems, and you like to talk. I think both approaches have its pros and cons, but note that @rafalcieslak used first approach, that's why he was so productive, most of his work is filled with TODOs, stubs, etc. but in the end the most important feature is done, and gaps to be filled are prepared, leaving some place for redesign, but... we have experience, useful code, and see potential problems.

In this case where neither of us is experienced with this type of project prototyping enables us to gain some experience, get something working to show to attract more people to project, speeds up things a lot, gives a feeling that creating something like this is possible and much more.

cahirwpz, 2017-06-06:
`vnode` operations, that are likely interesting to us, but missing in *Mimiker*:

```c
/* NetBSD7 */ int VOP_READLINK(vnode_t *v, uio_t *uio, kauth_cred_t cred);
/* NetBSD7 */ int VOP_SETATTR(vnode_t *v, vattr_t *va, kauth_cred_t cred);
/* NetBSD7 */ int VOP_ACCESS(struct vnode *vp, int mode, kauth_cred_t cred);
/* NetBSD7 */ int VOP_RENAME(vnode_t *fdv, vnode_t *fv, struct componentname *fcn,
                             vnode_t *tdv, vnode_t *tv, struct componentname *tcn);
/* NetBSD7 */ int VOP_SYMLINK(vnode_t *dv, vnode_t **vp, struct componentname *cnp,
                              vattr_t *vap, char *target);
```

pwit81, 2017-06-12:
There's a slight difference in semantics of `VOP_LOOKUP` between NetBSD7 and FreeBSD11 regarding objects locked on successfull execution.

`int VOP_LOOKUP(vnode_t *dv, vnode_t **vp, struct componentname *cnp);`

If the operation is successful *vp is unlocked (in NetBSD7) or locked (FreeBSD11). This requires more investigation.

cahirwpz, 2019-12-11:
We need all `vnodeops` and `vfsops` documented in _Mimiker_. At least we should note down facts about locking and memory ownership.

cahirwpz, 2020-03-31:
The interface has been largely improved by @xThaid within last months. Almost all parts of VFS was touched. Hence I think we can start tracking _vops_ and _vfsops_ improvements individually.

closed 2020-03-31 by cahirwpz