cahirwpz/mimiker #380: ISA bus driver

cahirwpz, 2017-06-10:
I think we completely missed the way to approach ISA bus handling so far. With [PnP ISA](http://www.osdever.net/documents/PNP-ISA-v1.0a.pdf) standard we can actually discover what devices are attached to ISA bus and what kind of hardware resources do they expose (interrupts, DMA channels, I/O ports). Here's the list of interesting ISA device drivers in FreeBSD kernel:

* [serial port](http://bxr.su/FreeBSD/sys/dev/uart/uart_bus_isa.c#63)
* [parallel port](http://bxr.su/FreeBSD/sys/dev/ppc/ppc_isa.c#87)
* [AT keyboard](http://bxr.su/FreeBSD/sys/dev/atkbdc/atkbdc_isa.c#88)
* [AT interrupt controller](http://bxr.su/FreeBSD/sys/powerpc/mpc85xx/atpic.c#115)
* [AT realtime clock](http://bxr.su/FreeBSD/sys/x86/isa/atrtc.c#221)
* [AT DMA controller](http://bxr.su/FreeBSD/sys/x86/isa/isa_dma.c#571)
* [AT timer](http://bxr.su/FreeBSD/sys/x86/isa/clock.c#628)

The work on ISA bus should be divided into many steps. We'll track them with following task list:

 * [ ] print all devices on ISA bus discovered with PnP protocol

pwit81, 2017-07-05:
QEMU doesn't seem to provide ISA PnP interface as demonstrated by [isa-pnp](https://github.com/cahirwpz/mimiker/tree/isa-pnp).

cahirwpz, 2019-02-14:
Detection of devices attached to ISA bus will be aided by Flat Device Tree hints.

cahirwpz, 2020-06-19:
Relevant FDT definition for [malta](https://github.com/cahirwpz/mimiker/blob/master/sys/mips/malta.dts).

mohrcore, 2021-02-08:
I guess this issue is no longer relevant. I'm not sure if qemu still doesn't provide an interface for ISA PnP devices, but what I am sure of is that we won't ever care to add PnP support for ISA, given how we are shifting our focus to RPi platforms.

WIth the ISA driver now merged into the master branch I don't think there's any special work to be done in regards to it. At least not any that goes beyond the scope of general driver infrastructure improvements and hacks that could be needed in the future to deal with the level of intertwining between the PCI and ISA buses wthin PIIX4.

Hence, I'm closing this issue.

closed 2021-02-08 by mohrcore