cahirwpz/mimiker #407: Can't pass arguments to user space programs

MoloToha, 2018-03-12:
Looks like currently there is no logic for passing arguments to user space programs. I accidentally discovered it while trying to run utest and being unable to pass a required argument to it

```
if (argc < 2) {
  printf("Not enough arguments provided to utest.\n");
  return 1;
}
```

This is caused by hard-coding program arguments in [sys/main.c](http://mimiker.ii.uni.wroc.pl/source/xref/mimiker/sys/main.c)

```
exec_args_t init_args = {
  .prog_name = init, .argc = 1, .argv = (const char *[]){init}};
```

MoloToha, 2018-03-12:
If you try to launch user space tests (`./launch init=/bin/utest`), the program will simply fail, because there in no way you can pass arguments to it (no documented way, at least)

cahirwpz, 2018-03-20:
What do you exactly mean? When I launch kernel with following command:

> ./launch init=/bin/ls_rec **arg1 arg2 arg3**

I get following message from the kernel:

> [malta.c:189] Switching to 'kernel-main' thread...
> Kernel arguments (10): mimiker.elf rd_start=0xffffffff80140000 rd_size=11949568 init=/bin/ls_rec **arg1 arg2 arg3** memsize=134217728 ememsize=134217728 modetty0=38400n8r

MoloToha, 2018-03-26:
@cahirwpz Your example shows that those arguments are passed to kernel, but they are not passed to a user-space program

dingorth, 2018-04-08:
@MoloToha In case you want to run single userspace test, you can do it with `./launch test=user_[test_name]` e.g. `./launch test=user_stat`. 

Running all userspace tests without kernel tests, requires multiple `/bin/utest` execution with different arguments. I think it can be done with some flag like "all_user" added to `ktest_main` in ktest.c.

dingorth, 2018-04-08:
One solution is to pass arguments like `./launch init=/bin/utest user_args="arg1 arg2"` and modify setup_kenv in mips/malta.c. We need to distinguish userspace program arguments and kernel arguments thus key-value syntax with user_args may be usefull.

cahirwpz, 2019-11-13:
Done in #582

closed 2019-11-13 by cahirwpz