cahirwpz/mimiker #516: Adding member to `struct proc` breaks tests

staffik, 2019-01-08:
Tests failed when added new member `LIST_ENTRY(proc) dummy` to `struct proc` and runned `./run_tests.py`.

jpszczolowski, 2019-01-09:
@staffik did you get the following kernel panic?
```
# Running test "user_fork_signal".
WARNING: This test will enter usermode.
Starting user test "fork_signal".
[/home/mimiker/mimiker/sys/mutex.c:23] PANIC: Sleeping mutex 0x80dab190 is not recursive!

Reading symbols from mimiker.elf...done.
panic_fail () at /home/mimiker/mimiker/sys/assert.c:13
13	    ;
Breakpoint 1 at 0x8011a2b0: file /home/mimiker/mimiker/sys/startup.c, line 20.
Breakpoint 2 at 0x8010d7d0: file /home/mimiker/mimiker/sys/assert.c, line 16.
Breakpoint 3 at 0x8010d7b8: file /home/mimiker/mimiker/sys/assert.c, line 6.
(gdb) backtrace
#0  panic_fail () at /home/mimiker/mimiker/sys/assert.c:13
#1  0x801143c8 in _mtx_lock (m=0x80dab190, waitpt=0x80117040 <proc_exit+36>, waitpt@entry=0x80116e20 <proc_lock+12>) at /home/mimiker/mimiker/sys/mutex.c:23
#2  0x80116e28 in mtx_lock (m=<optimized out>) at /home/mimiker/mimiker/include/mutex.h:58
#3  proc_lock (p=<optimized out>) at /home/mimiker/mimiker/sys/proc.c:51
#4  0x80117048 in proc_exit (exitstatus=0x2) at /home/mimiker/mimiker/sys/proc.c:133
#5  0x80119238 in sig_exit (td=<optimized out>, sig=SIGILL) at /home/mimiker/mimiker/sys/signal.c:174
#6  0x8010a35c in sig_send (sig=SIGCHLD, sa=0x80dab218) at /home/mimiker/mimiker/mips/signal.c:42
#7  0x80119314 in sig_post (sig=SIGCHLD) at /home/mimiker/mimiker/sys/signal.c:168
#8  0x8010f528 in on_user_exc_leave () at /home/mimiker/mimiker/sys/exception.c:31
#9  0x80108e08 in mips_exc_handler (frame=0x800f4ef0) at /home/mimiker/mimiker/mips/intr.c:298
#10 0x80103198 in user_exc_enter () at /home/mimiker/mimiker/mips/exc.S:81
Backtrace stopped: frame did not save the PC
(gdb) kthread
(*) current thread marker
-------------------------------------------------------------------------------------------------
|     Id | Name              | State        | Priority | Waiting Point                          |
-------------------------------------------------------------------------------------------------
|      0 | idle-thread       | TDS_READY    |      255 | null                                   |
|      1 | callout           | TDS_SLEEPING |       32 | /home/mimiker/mimiker/sys/sleepq.c:342 |
|      2 | main              | TDS_SLEEPING |       32 | /home/mimiker/mimiker/sys/proc.c:241   |
| (*) 66 | utest-fork_signal | TDS_RUNNING  |       32 | null                                   |
|     67 | utest-fork_signal | TDS_DEAD     |       32 | null                                   |
-------------------------------------------------------------------------------------------------
```

jpszczolowski, 2019-01-09:
I've got another kernel panic. I've added additional `mtx_t some_unused_lock;` at the beginning of `struct proc` and now, after `make && make test` I get:
```
(...)
  user_fd_all  <---- FAILED
The seed used for this test order was: 0. Start kernel with `test=all seed=0 repeat=1` to reproduce this test case.
[/home/mimiker/mimiker/sys/ktest.c:67] PANIC: Halting kernel on failed test.

Reading symbols from mimiker.elf...done.
panic_fail () at /home/mimiker/mimiker/sys/assert.c:13
13	    ;
Breakpoint 1 at 0x8011a2b0: file /home/mimiker/mimiker/sys/startup.c, line 20.
Breakpoint 2 at 0x8010d7d0: file /home/mimiker/mimiker/sys/assert.c, line 16.
Breakpoint 3 at 0x8010d7b8: file /home/mimiker/mimiker/sys/assert.c, line 6.
(gdb) backtrace
#0  panic_fail () at /home/mimiker/mimiker/sys/assert.c:13
#1  0x8011327c in ktest_failure () at /home/mimiker/mimiker/sys/ktest.c:67
#2  0x80108b68 in kernel_oops (frame=0x800f4c2c) at /home/mimiker/mimiker/mips/intr.c:255
#3  0x80108cb0 in ade_handler (frame=<optimized out>) at /home/mimiker/mimiker/mips/intr.c:196
#4  0x80108d6c in mips_exc_handler (frame=0x800f4c2c) at /home/mimiker/mimiker/mips/intr.c:287
#5  0x80103404 in kern_exc_enter () at /home/mimiker/mimiker/mips/exc.S:172
#6  0x8011f714 in __preempt_disable (data=0x0) at /home/mimiker/mimiker/include/sched.h:28
#7  vm_map_activate (map=0x1) at /home/mimiker/mimiker/sys/vm_map.c:35
#8  0x8010b3b8 in ctx_switch () at /home/mimiker/mimiker/mips/switch.S:70
Backtrace stopped: frame did not save the PC
(gdb) kthread
(*) current thread marker
----------------------------------------------------------------------------------------------
|     Id | Name         | State        | Priority | Waiting Point                            |
----------------------------------------------------------------------------------------------
|      0 | idle-thread  | TDS_READY    |      255 | null                                     |
|      1 | callout      | TDS_SLEEPING |       32 | /home/mimiker/mimiker/sys/sleepq.c:342   |
|      2 | main         | TDS_SLEEPING |       32 | /home/mimiker/mimiker/sys/proc.c:242     |
| (*) 52 | utest-fd_all | TDS_RUNNING  |       32 | null                                     |
|     53 | utest-fd_all | TDS_BLOCKED  |       32 | /home/mimiker/mimiker/sys/vm_object.c:31 |
----------------------------------------------------------------------------------------------
```
The problem here is that the argument `map` of `vm_map_activate(vm_map_t *map)` is equal to `0x1`.

cahirwpz, 2019-01-11:
This is a build system problem. Dependencies for `mips/assym.h` are not calculated so `P_USPACE` value is not updated after process structure is changed. This bug affects all structures referenced in `mips/genassym.c` file.

cahirwpz, 2019-01-11:
FTTB Please remove `mips/assym.h` whenever you modify following files:
```c
#include <thread.h>
#include <proc.h>
```
and other mentioned in `mips/genassym.c`.