cahirwpz/mimiker #518: Incorrect path resolution for '/bin/ls_rec' program

Tomatosoup97, 2019-01-09:
Running `/bin/ls_rec` fails, the reason for that seems to be invalid path resolution. One can see in the log below, that the script tries to run `Lua` using `/bin/ls_rec/bin/lua`, while it should be `/bin/lua`.

```
mu@mimiker ~/mimiker » ./launch init=/bin/ls_rec
(...)
[syscall] open("/lib/lua/init.lua", 0, 438)
[syscall] fstat(3, 0x7f7ff868)
[syscall] read(3, 0x08006590, 1024)
[syscall] close(3)
[syscall] open("/bin/ls_rec/bin/lua", 0, 438)
[syscall] write(2, 0x7f7fffdc, 8)
/bin/lua[syscall] write(2, 0x0044f002, 2)
: [syscall] write(2, 0x080065d0, 58)
cannot open /bin/ls_rec/bin/lua: No such file or directory[syscall] write(2, 0x0044f1ae, 1)
[syscall] exit(1)
[proc] Freeing process PID(0) {0x80dab028} resources
(...)
[/home/mimiker/mimiker/sys/proc.c:169] PANIC: 'init' process died!
```

cahirwpz, 2019-01-09:
`/bin/ls_rec` works well when launched through `/bin/mish`. This problem is most likely related to launching programs from kernel-space through `run_program` procedure.

cahirwpz, 2019-01-11:
Hmm... I was wrongly assuming that `strlcpy` returns the length of string including `NUL` character. That caused `argv` to be incorrectly copied into `exec_args_t` structure.

@mu It had nothing to do with path resolution – namely `vfs_lookup` routine.

commit c0a57cdc (2019-01-11): execve from kernel space works for shebang programs (fix #518)

 * strlcpy returns the length of copied string without NUL
closed 2019-01-11 by cahirwpz