cahirwpz/mimiker #567: Review signal handling routines and add missing functionality

cahirwpz, 2019-10-21:
[signal(7)](https://netbsd.gw.com/cgi-bin/man-cgi?signal+7) implementation in described in great detail in chapter 4.7 of _"The Design and Implementation of FreeBSD Operating System"_. We know that our implementation of signals is very rudimentary.

We care mainly about following system calls:
* [sigaction](https://netbsd.gw.com/cgi-bin/man-cgi?sigaction)
* `sigreturn`
* [sigaltstack](https://netbsd.gw.com/cgi-bin/man-cgi?sigaltstack)
* [sigprocmask](https://netbsd.gw.com/cgi-bin/man-cgi?sigprocmask)
* [sigsuspend](https://netbsd.gw.com/cgi-bin/man-cgi?sigsuspend)

`ksh` makes heavy use of `sigprocmask`, `sigsuspend` and `sigaction`. There're also important users of these syscalls in _libc_.

* [ ] remove `sa_restorer` from `struct sigaction` (kernel should inject `sigcode` that calls `sigreturn` onto userspace stack)
* [ ] support for `sigaction.sa_mask` (used in `ksh`)
* [ ] minimal support for `sigaction.sa_flags` (`SA_INTERRUPT` for `ksh` and `SA_RESTART` for `find`)
* [ ] handle `SIGQUIT`, `SIGHUP`
* [ ] handle `SIGTSTP`, `SIGTTIN`, `SIGTTOU`
* [ ] handle `SIGCONT`
* [ ] handle `SIGPIPE`

IIRC we don't handle `siginfo_t *` argument of `sa_sigaction` at all.

j-piecuch, 2020-05-20:
`SA_INTERRUPT` isn't defined in POSIX because the default behavior of signals is to interrupt system calls. Under SunOS, system calls were automatically restarted, and this flag disabled that behavior: https://www.tldp.org/LDP/lpg/node140.html

cahirwpz, 2020-05-26:
You're right – `SA_INTERRUPT` is not even defined in _NetBSD_ or _FreeBSD_ header files.

cahirwpz, 2022-11-22:
Since @j-piecuch implemented most of the features I consider this issue eligible for closure.

closed 2022-11-22 by cahirwpz