cahirwpz/mimiker #613: callout_simple test fails occasionaly

cahirwpz, 2019-12-03:
Very rarely `callout_simple` test fails. Here're messages extracted from CircleCI build:

```
...
# Running test "user_fd_bad_desc".
WARNING: This test will enter usermode.
Starting user test "fd_bad_desc".
# Running test "callout_stop".
# Running test "callout_simple".
Assertion "((elem)->c_flags & 0x0001)" at [/root/mimiker/sys/kern/callout.c:61] failed!
...
```

```
(gdb) backtrace full
#0  panic_fail () at /root/mimiker/sys/kern/assert.c:13
No locals.
#1  0xc011d910 in ktest_failure () at /root/mimiker/sys/kern/ktest.c:65
No locals.
#2  0xc0126340 in assert_fail (expr=expr@entry=0xc0139090 "((elem)->c_flags & 0x0001)", file=file@entry=0xc0138fec "/root/mimiker/sys/kern/callout.c", line=line@entry=61) at /root/mimiker/sys/kern/assert.c:19
No locals.
#3  0xc0125c1c in callout_thread (arg=<optimized out>) at /root/mimiker/sys/kern/callout.c:61
        elem = 0xc0024e10
#4  0xc011254c in thread_self () at /root/mimiker/sys/kern/thread.c:119
No locals.
(gdb) kthread
(*) current thread marker
+-------+-------------------+--------------+----------+-------------------+
|  Id   |       Name        |    State     | Priority |   Waiting Point   |
+=======+===================+==============+==========+===================+
|     0 |       idle-thread |    TDS_READY |      255 | null              |
|     1 |         interrupt | TDS_SLEEPING |        0 | kern/sleepq.c:346 |
| (*) 2 |           callout |  TDS_RUNNING |       32 | null              |
|     3 |              main | TDS_SLEEPING |       32 | kern/sleepq.c:346 |
|    87 | utest-fd_bad_desc |     TDS_DEAD |       32 | null              |
+-------+-------------------+--------------+----------+-------------------+
```

jpszczolowski, 2019-12-08:
@cahirwpz have you saved the test seed?

cahirwpz, 2019-12-11:
Unfortunately not.

cahirwpz, 2019-12-11:
This could have been a byproduct of recently fixed problem with `tlb_refill`. I cannot observe it anymore.

closed 2019-12-11 by cahirwpz