cahirwpz/mimiker #672: Long command in ksh causes kernel fail

jpszczolowski, 2020-04-10:
Steps to reproduce:
* Launch `/bin/ksh`
* Enter a long command like this (512 letters `a`):
```
aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
```
* Kernel fails:
```
(gdb) bt
#0  assert_fail (expr=expr@entry=0xc0137780 "state->compare.val > state->count.val", file=file@entry=0xc0137754 "/home/jpszczolowski/mim2/sys/mips/timer.c", line=line@entry=0x45) at /home/jpszczolowski/mim2/sys/kern/assert.c:16                          
#1  0xc0102fd4 in set_next_tick (state=0xc0140010 <mips_timer_state>) at /home/jpszczolowski/mim2/sys/mips/timer.c:69
#2  0xc0103000 in mips_timer_intr (data=0xc0140048 <mips_timer>) at /home/jpszczolowski/mim2/sys/mips/timer.c:79
#3  0xc0121cfc in intr_event_run_handlers (ie=ie@entry=0xc0143200 <mips_intr_event+392>) at /home/jpszczolowski/mim2/sys/kern/interrupt.c:91                                                                                                                 
#4  0xc0105724 in mips_intr_handler (frame=<optimized out>) at /home/jpszczolowski/mim2/sys/mips/interrupt.c:73
#5  0xc0102abc in mips_exc_handler (frame=0xc0025e14) at /home/jpszczolowski/mim2/sys/mips/trap.c:234
#6  0x801009a8 in kern_exc_enter () at /home/jpszczolowski/mim2/sys/mips/ebase.S:483
#7  0xc01219a4 in intr_enable () at /home/jpszczolowski/mim2/sys/kern/interrupt.c:28
#8  0x0000000c in ?? ()
```

mzr, 2020-05-04:
Failed to reproduce on c92f35be2aeb15c70ff207c1e7a5225e7a66655b
https://asciinema.org/a/SHRXxpVx7xlByQ2PNKEdnEXUi

jpszczolowski, 2020-05-05:
@mzr I still see this panic on master branch. Maybe try longer command?