cahirwpz/mimiker #685: Use-after-free in device_add_resource

jpszczolowski, 2020-04-22:
**Error details**
```
===========KernelAddressSanitizer===========
ERROR:
* invalid access to address 0xc0014250
* write of size 4
* redzone code 0xfd (pool use-after-free)
============================================
```
```
#0  panic_fail () at /home/jpszczolowski/mimiker/sys/kern/assert.c:6
#1  0xc010e618 in kasan_shadow_check (read=0x0, size=0x4, addr=addr@entry=0xc0014250) at /home/jpszczolowski/mimiker/sys/kern/kasan.c:180
#2  __asan_store4_noabort (addr=addr@entry=0xc0014250) at /home/jpszczolowski/mimiker/sys/kern/kasan.c:244
#3  0xc013fc88 in device_add_resource (dev=dev@entry=0xc0024b58, r=r@entry=0xc0014268, rid=rid@entry=0x0) at /home/jpszczolowski/mimiker/sys/kern/device.c:60
#4  0xc010777c in gt_pci_alloc_resource (pcib=<optimized out>, dev=0xc0024b58, type=RT_ISA, rid=0x0, start=0x60, end=0x67, size=0x8, flags=RF_ACTIVE) at /home/jpszczolowski/mimiker/sys/mips/gt64120.c:362
#5  0xc010cbd4 in bus_alloc_resource (flags=RF_ACTIVE, size=0x8, end=0x67, start=0x60, rid=0x0, type=RT_ISA, dev=0xc0024b58) at /home/jpszczolowski/mimiker/include/sys/bus.h:181
#6  atkbdc_attach (dev=0xc0024b58) at /home/jpszczolowski/mimiker/sys/drv/atkbdc.c:162
#7  0xc013fa6c in device_attach (dev=dev@entry=0xc0024b58) at /home/jpszczolowski/mimiker/sys/kern/device.c:37
#8  0xc013fc04 in make_device (parent=<optimized out>, driver=driver@entry=0xc0171c40 <atkbdc_driver>) at /home/jpszczolowski/mimiker/sys/kern/device.c:53
#9  0xc010ca94 in atkbdc_init () at /home/jpszczolowski/mimiker/sys/drv/atkbdc.c:191
#10 0xc0123c54 in sysinit () at /home/jpszczolowski/mimiker/sys/kern/sysinit.c:89
#11 0xc0127d0c in kernel_init () at /home/jpszczolowski/mimiker/sys/kern/startup.c:33
#12 0xc0106440 in platform_init () at /home/jpszczolowski/mimiker/sys/mips/malta.c:171
#13 0x80101044 in _start () at /home/jpszczolowski/mimiker/sys/mips/start.S:43
```
Error is in this line, while trying to access `dev->resources.tqh_first`:
```C
TAILQ_INSERT_HEAD(&dev->resources, r, r_device);
```
**What happened (tl;dr)**
`atkbdc_probe` allocates a new resource and adds it to `dev->resources` here:
```
#0  0xc013fcd8 in device_add_resource (dev=dev@entry=0xc0024b58, r=r@entry=0xc0014220, rid=rid@entry=0x0) at /home/jpszczolowski/mimiker/sys/kern/device.c:60
#1  0xc010777c in gt_pci_alloc_resource (pcib=<optimized out>, dev=0xc0024b58, type=RT_ISA, rid=0x0, start=0x60, end=0x67, size=0x8, flags=RF_ACTIVE) at /home/jpszczolowski/mimiker/sys/mips/gt64120.c:362
#2  0xc010cf94 in bus_alloc_resource (flags=RF_ACTIVE, size=0x8, end=0x67, start=0x60, rid=0x0, type=RT_ISA, dev=0xc0024b58) at /home/jpszczolowski/mimiker/include/sys/bus.h:181
#3  atkbdc_probe (dev=0xc0024b58) at /home/jpszczolowski/mimiker/sys/drv/atkbdc.c:127
#4  0xc013f96c in device_probe (dev=dev@entry=0xc0024b58) at /home/jpszczolowski/mimiker/sys/kern/device.c:28
#5  0xc013fbd4 in make_device (parent=<optimized out>, driver=driver@entry=0xc0171c40 <atkbdc_driver>) at /home/jpszczolowski/mimiker/sys/kern/device.c:52
#6  0xc010ca94 in atkbdc_init () at /home/jpszczolowski/mimiker/sys/drv/atkbdc.c:191
#7  0xc0123c54 in sysinit () at /home/jpszczolowski/mimiker/sys/kern/sysinit.c:89
#8  0xc0127d0c in kernel_init () at /home/jpszczolowski/mimiker/sys/kern/startup.c:33
#9  0xc0106440 in platform_init () at /home/jpszczolowski/mimiker/sys/mips/malta.c:171
#10 0x80101044 in _start () at /home/jpszczolowski/mimiker/sys/mips/start.S:43
```
Then `atkbdc_probe` deallocates the resource here:
```
#0  pool_free (pool=0xc0190438 <P_POOL_BOOTPAGE+1080>, ptr=ptr@entry=0xc0014220) at /home/jpszczolowski/mimiker/sys/kern/pool.c:239
#1  0xc012ca70 in rman_release_resource (r=r@entry=0xc0014220) at /home/jpszczolowski/mimiker/sys/kern/rman.c:99
#2  0xc0107538 in gt_pci_release_resource (pcib=<optimized out>, dev=<optimized out>, type=<optimized out>, rid=<optimized out>, r=0xc0014220) at /home/jpszczolowski/mimiker/sys/mips/gt64120.c:368
#3  0xc010d110 in bus_release_resource (r=0xc0014220, rid=0x0, type=RT_ISA, dev=0xc0024b58) at /home/jpszczolowski/mimiker/include/sys/bus.h:217
#4  atkbdc_probe (dev=0xc0024b58) at /home/jpszczolowski/mimiker/sys/drv/atkbdc.c:147
#5  0xc013f96c in device_probe (dev=dev@entry=0xc0024b58) at /home/jpszczolowski/mimiker/sys/kern/device.c:28
#6  0xc013fbd4 in make_device (parent=<optimized out>, driver=driver@entry=0xc0171c40 <atkbdc_driver>) at /home/jpszczolowski/mimiker/sys/kern/device.c:52
#7  0xc010ca94 in atkbdc_init () at /home/jpszczolowski/mimiker/sys/drv/atkbdc.c:191
#8  0xc0123c54 in sysinit () at /home/jpszczolowski/mimiker/sys/kern/sysinit.c:89
#9  0xc0127d0c in kernel_init () at /home/jpszczolowski/mimiker/sys/kern/startup.c:33
#10 0xc0106440 in platform_init () at /home/jpszczolowski/mimiker/sys/mips/malta.c:171
#11 0x80101044 in _start () at /home/jpszczolowski/mimiker/sys/mips/start.S:43
```
but the resource is still on `dev->resources` tail queue!

Later, within `device_attach`, we access the freed resource while trying to add another resource to `dev->resources`.

jpszczolowski, 2020-04-22:
My first thought: as `gt_pci_alloc_resource` calls `device_add_resource`, maybe `gt_pci_release_resource` should call something like `device_release_resource` (such function does not exist)?.

mzr, 2020-04-22:
Probably that's the reason: https://github.com/cahirwpz/mimiker/blob/master/sys/drv/atkbdc.c#L125

PR #698 (2020-05-01): Remove resource from device before deallocation
* `gt_pci_release_resource` now calls `device_remove_resource`, so deallocated resource does not stay on `dev->resources` tail queue (note that `gt_pci_alloc_resource` calls `device_add_resource`, which adds a newly allocated resource to `dev->resources`)
* `device_alloc` now calls `TAILQ_INIT(&dev->resources)` (another bug fixed?)

Fixes #685
closed 2020-05-01 by cahirwpz