cahirwpz/mimiker #1319: Copying memory seems to be unreliable.

mohrcore, 2022-07-24:
I'm facing a bizarre issue with copying memory. Currently [this version of GPIO driver code](https://github.com/cahirwpz/mimiker/pull/1277/commits/83b0d0de60855d9cc5fb8a12da9612842af67d51) won't print the name of an entry on error. The string is being recognized as empty. However, after removing the `function` property from `emmc` FDT node, and placing a breakpoint on `bcm2835_gpio.c:114`, the name gets printed as it should. The `FDT_getprop` function is what's used to retrieve the name string. After placing `klog` calls inside it, it looks like `buf` does not contain a copy of `prop` after calling `memcpy`. Weirdly enough, hitting a breakpoint, seems to synchronize memory somehow and this effect is gone.

I'm guessing this might be related to the recent bump of QEMU.

pj1031999, 2022-07-25:
@mohrcore could you provide step-by-step scenario?
```
./lunch --something
b file:line
c
p var
-> what you see
-> what should be seen
```

cahirwpz, 2022-07-25:
Have you run your code with KASAN enabled? How exactly can we reproduce the bug?

mohrcore, 2022-07-25:
@pj1031999 
> @mohrcore could you provide step-by-step scenario?
> 
> ```
> ./lunch --something
> b file:line
> c
> p var
> -> what you see
> -> what should be seen
> ```

Sure.

First, you need to remove line no. 65 from `sys/dts/rpi3.dts`. This will cause the node to become an invalid GPIO configuration entry and will trigger the code inside the block that starts in `sys/drv/bcm2835_gpio.c:113`.

Launch mimiker with
```
./launch -b rpi3 -d init=/bin/ksh
```

In gdb continue, then pause and check klog.
```
(gdb) c
^C
(gdb) klog
```

The expected `klog` output should contain the following line:
```
Warning: GPIO FDT entry "emmc" with no `function` property
```

Instead we get
```
Warning: GPIO FDT entry "" with no `function` property
```

However, if we place a breakpoint on line `sys/drv/bcm2835.cL114`, we can `print` the entry name that's about to be printed to klog and see that now it contains `"emmc"`. After progressing with `next` we can check the `klog` and indeed it prints the expected string.

```
(gdb) b bcm2835_gpio.c:114
(gdb) c
# We should hit the breakpoint now!
(gdb) print entry_name
(gdb) n
(gdb) klog
```

mohrcore, 2022-07-25:
@cahirwpz I'm having some trouble compiling mimiker with KASAN enabled.
I'm doing
```
make distclean
make BOARD=rpi3 KASAN=1
```
`make` complains about no rules for building `include/machine/cdefs.h` that's required by `kasan_quar.o`.

I'll try later on a fresh repo and see whether it helps.

pj1031999, 2022-07-25:
@mohrcore you allocate memory for name, copy that name into buffer, pass *pointer* to klog and free that buffer. I guess that later other call to kmalloc returns that memory and it's zeroed so as a result you see empty string.

So... use-after-free :)

closed 2022-07-25 by pj1031999